warpem/warp #323: Unable to access http://www.warpem.com/

abnerabner, 2024-12-22:
Hi, 

Recently I tried to use Warp on Windows, but I kept failing to access `http://www.warpem.com/` with the error DNS_PROBE_FINISHED_NXDOMAIN. I have tried on several computers, but I can't access it normally.

Do you have any suggestions?

Thanks!

Best,
Abner

alisterburt, 2024-12-22:
Hi @abnerabner - sorry about this, Dimitry lost access to the server that site was run from. 

We don't currently have a system in place for providing windows builds and windows warp is not actively maintained right now - I would suggest asking on the google group for a v1.10 installer if you still want to try it out, otherwise trying WarpTools

abnerabner, 2024-12-23:
Got it. Thank you

closed 2024-12-23 by abnerabner