warpem/warp #325: How to access Particle Picking Models

diogojsf, 2025-01-06:
Dear colleagues,
I'm planning on installing Warp but whenever I try to access the pre-picking models from this webpage https://warpem.github.io/warp/user_guide/warp/boxnet/retraining/ to the link attached to the "the training data repository" the page won't open.
[image]

Will this create any issue in the installation or running the boxnet particle picking?

Diogo Ferreira

alisterburt, 2025-01-06:
Hey @diogojsf - sorry about that, the old website is out of action. In general desktop Warp is no longer actively supported and we are prioritising the linux version.

BoxNet is one of the last things that needs finalizing in the linux versions of everything and is high priority for this year (c.f. https://github.com/warpem/warp/issues/14#issuecomment-2536836542)

alisterburt, 2025-01-07:
@diogojsf are you asking about in the desktop app or for the linux tools? 

Training in the desktop app works but is actually retraining based on a given BoxNet model so you would need the weights for that. Training on linux from the CLI is not supported yet.

diogojsf, 2025-01-09:
Linux tools!

So I'm planning on instaling Warp on our workstation but I'd like to take advantage of boxnet and take the particles for 3d. Could you give me some advice on how to accomplish this? The workstation is linux based, so, to accomplish this I would need to install Warp on windows right?

alisterburt, 2025-01-09:
@diogojsf BoxNet is not a 3D particle picker - it sounds like you're not clear on the Warp workflow/what programs are available, I recommend working through the tutorial https://warpem.github.io/warp/user_guide/warptools/quick_start_warptools_tilt_series/