warpem/warp #326: Memory leak? fs_motion_and_ctf accumulates RAM

dmichalak, 2025-01-09:
Using the follow commands, I set up and ran motion and ctf processing on our data where the RAM was taken up until the workstation froze. The data set contains 969 frame series, 762 were processed before crashing. 

```
WarpTools create_settings \
	--folder_data frames \
	--folder_processing warp_frameseries \
	--output warp_frameseries.settings \
	--extension "*.tif" \
	--angpix 1.091 \
	--exposure 2.4

WarpTools create_settings \
	--output warp_tiltseries.settings \
	--folder_processing warp_tiltseries \
	--folder_data tomostar \
	--extension *.tomostar \
	--angpix 1.091 \
	--exposure 2.4 \
	--tomo_dimensions 4400x6000x1000

WarpTools fs_motion_and_ctf \
	--settings warp_frameseries.settings \
	--m_grid 1x1x3 \
	--c_grid 2x2x1 \
	--c_range_max 7 \
	--c_defocus_max 10 \
	--c_use_sum \
	--out_averages \
	--out_average_halves \
	--perdevice 2
```

System specs are
Ubuntu 22.04.5 LTS
96x Xeon Gold 5220R CPU 
4x RTX 3090
512 GB memory

alisterburt, 2025-01-09:
thanks for the report @dmichalak and sorry you ran into issues! I'll try to take a look next week, in the meantime you might have to work with smaller subsets of your data ([docs](https://warpem.github.io/warp/reference/warptools/subsets/)) if it is indeed a memory leak

alisterburt, 2025-01-09:
Processing 287 frame series from the tutorial data I see fairly constant memory usage around the 5GB mark and no obvious linear increase per item processed... these are also tiffs - any ideas what's different about this data? Do you see this issue with other datasets/the tutorial data?

dmichalak, 2025-01-09:
The linear increase per item also occurs with the tutorial dataset. It seems to increase roughly 400-800 MB per item when using `--device_list 0`. Removing and re-creating the conda environment doesn't change anything. I'm somewhat at a loss for what could be the issue.

Processing one tilt series at a time seems to be the way to go in the meantime.

alisterburt, 2025-01-09:
I'm not sure what could be going on here... without being able to reproduce there isn't much I can do unfortunately

alisterburt, 2025-01-15:
@dmichalak test worth running - if you set `CUDA_VISIBLE_DEVICES` manually rather than passing `--device_list` does the behavior change?

dmichalak, 2025-01-21:
Running the script below on the tutorial data did not change the behavior. I wonder if building warp locally might change anything?

```
export CUDA_VISIBLE_DEVICES=0

WarpTools fs_motion_and_ctf \
	--settings warp_frameseries.settings \
	--m_grid 1x1x3 \
	--c_grid 2x2x1 \
	--c_range_max 7 \
	--c_defocus_max 8 \
	--c_use_sum \
	--out_averages \
	--out_average_halves \
	--output_processing motion_and_ctf \

```

dmichalak, 2025-01-21:
@alisterburt Using `perdevice=1` seems to be a workaround. The RAM fills up for each process until a certain point, approx. 16% of total RAM per process for a total of 60% (4 GPUs, 4 devices). Then eventually the processes empty and refill up to around 16%. Having `perdevice=2` makes 8 processes and they can't all go up to 16%. 

It seems like the program looks at the total amount of RAM and tells each process to use up to a certain amount based on that total.

dtegunov, 2025-01-22:
Hey! Sounds like C#'s garbage collector isn't feeling enough pressure to collect freed up memory before it's too late. I don't know if it has some heuristics internally and 512 GB makes it think "I'll never run out of so much memory, why even bother." Can you please try setting the DOTNET_GCHeapHardLimit environment variable before running the application, setting it to the amount of memory you'd like each worker process to stay under, in MB? For instance,
`export DOTNET_GCHeapHardLimit=16384`
should make it stay within 16 GB (if the amount of data that needs to be held in memory at any given time is less than that).

dmichalak, 2025-01-23:
Hello! 

Adding `export DOTNET_GCHeapHardLimit=16384` causes an error.
```
GC heap initialization failed with error 0x8007000E
Failed to create CoreCLR, HRESULT: 0x8007000E
```