warpem/warp #333: MTools create_source fails when population directory is outside of the parent warp directory

jmdobbs, 2025-01-11:
Apparently due to relative path problems somewhere.

M seems to be looking for the tomostar folder in the current directory only, not in the directory where the settings file is

'''
[dobbs@srv-mahamid-01: /scratch/dobbs/cells/untreated/relion] $  MTools create_source --name m1_ribo_source --population m1/m1_ribo.population --processing_settings ../20241220_first_bw/settings_tomos.settings  
Running command create_source with:
population = m1/m1_ribo.population
processing_settings = ../20241220_first_bw/settings_tomos.settings
name = m1_ribo_source
nframes = 
files = 

Looking for data... Unhandled exception. System.IO.DirectoryNotFoundException: Could not find a part of the path '/scratch/dobbs/cells/untreated/relion/metadata_tomostar'.
   at System.IO.Enumeration.FileSystemEnumerator`1.CreateDirectoryHandle(String path, Boolean ignoreNotFound)
   at System.IO.Enumeration.FileSystemEnumerator`1.Init()
   at System.IO.Enumeration.FileSystemEnumerable`1..ctor(String directory, FindTransform transform, EnumerationOptions options, Boolean isNormalized)
   at System.IO.Enumeration.FileSystemEnumerableFactory.UserFiles(String directory, String expression, EnumerationOptions options)
   at System.IO.Directory.InternalEnumeratePaths(String path, String searchPattern, SearchTarget searchTarget, EnumerationOptions options)
   at MTools.Commands.CreateSource.Run(Object options)
   at MTools.MTools.Run(Object options)
   at CommandLine.ParserResultExtensions.WithParsed[T](ParserResult`1 result, Action`1 action)
   at MTools.MTools.Main(String[] args)
Aborted (core dumped)

'''

alisterburt, 2025-01-13:
Hey @jmdobbs - thanks for the report! good to know - not a priority to fix but will keep it in mind

jmdobbs, 2025-01-16:
accidental duplicate of #312

closed 2025-01-16 by jmdobbs