warpem/warp #334: Failed to process data marked as unselected

CleoShen, 2025-01-13:
Hi all, I'm running the Warp2 on a computer cluster node. And the "Failed to process data marked as unselected" error happens both on the frame data and TS data. And I've tried to use the change_selection command to mark them as "selected", but the error still poped-up.
More specifically, I'm running the command of `WarpTools fs_ctf --settings warp_frameseries.settings --grid 2x2x1 --range_max 7 --defocus_max 8 --use_sum`, and notified me that `Failed to process ..../warp_frameseries/Position_1_030_45.00_EER.eer, marked as unselected Check logs in .../warp_frameseries/logs for more info. Use the change_selection WarpTool to reactivate this item if required.` Then I used the `WarpTools change_selection --settings warp_frameseries.settings --select` and ran the ts_ctf command again, it returns the same error.

alisterburt, 2025-01-13:
@CleoShen did you look in the worker logs as suggested? What did they say? 

> Check logs in .../warp_frameseries/logs for more info.

CleoShen, 2025-01-13:
This is one of my frame's log file, and I've used apostrophes to replace some of the pathway:
`2025-01-13 20:37:28.205 Received "LoadStack", with 4 arguments, for GPU #0, 5856 MB free:
2025-01-13 20:37:33.504 System.Exception: Could not successfully read ....../warp_frameseries/average/Position_1_001_0.00_20241128_173728_EER.mrc within the specified number of attempts.
2025-01-13 20:37:33.504    at Warp.Headers.MapHeader.ReadFromFilePatient(Int32 attempts, Int32 mswait, String path, int2 headerlessSliceDims, Int64 headerlessOffset, Type headerlessType) in ....../programs/run/warp-2.0.0dev26/WarpLib/Headers/Headers.cs:line 127
2025-01-13 20:37:33.504    at WarpWorker.WarpWorker.LoadAndPrepareStack(String path, Decimal scaleFactor, Boolean correctGain, Int32 maxThreads) in ....../programs/run/warp-2.0.0dev26/WarpWorker/WarpWorker.cs:line 824
2025-01-13 20:37:33.504    at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in ....../programs/run/warp-2.0.0dev26/WarpWorker/WarpWorker.cs:line 227`

alisterburt, 2025-01-14:
@CleoShen warp expects frame series data, you have to trick it to use data that has already been motion corrected. See https://warpem.com/reference/warptools/processing_tilt_series_stacks/ for some details/tips

CleoShen, 2025-01-14:
@alisterburt Thank you for your fast response. But this site can't be reached.

alisterburt, 2025-01-14:
My bad https://warpem.com/reference/warptools/processing_tilt_series_stacks/

CleoShen, 2025-01-14:
For the specific case I mentioned above was because I created the frame setting file with EER extension, but I skipped the motion correction step, which locked the frames. After changing the frame setting with MRC extension, that error was cleared. 
But in my second case, basically, I followed the tutorial of "Quick Start: Tilt Series", the unselected error poped-up at the step of ts_stack, do you have any idea? Here are all the commands I ran:
1.  $ WarpTools create_settings --folder_data ../frames/warptest2 --gain_path ../frames/CountRef_241128_TEM4.mrc --folder_processing warp_frameseries --output warp_frameseries.settings --extension "*.eer" --angpix 1.240 --exposure 3.34 --eer_ngroups 15 --gain_flip_y
2. $ WarpTools fs_motion_and_ctf --settings warp_frameseries.settings --m_grid 1x1x10 --c_grid 2x2x1 --c_range_max 7 --c_defocus_max 9 --c_use_sum --out_averages
3. $ WarpTools ts_import --mdocs ../mdoc/mrc_mdoc/warptest2 --frameseries warp_frameseries --tilt_exposure 3.34 --min_intensity 0 --dont_invert --output tomostar
4. $ WarpTools create_settings --output warp_tiltseries.settings --folder_processing warp_tiltseries --folder_data tomostar --extension "*.tomostar" --angpix 1.24 --tomo_dimensions 4096x4096x1500 --exposure 3.34
5. $ WarpTools ts_aretomo --settings warp_tiltseries.settings --angpix 4.96 --alignz 1500 --patches 4x4 --exe AreTomo_1.3.4_2023

alisterburt, 2025-01-14:
@CleoShen as I suggested in the first message, you will need to check the worker logs

CleoShen, 2025-01-15:
Following up on my second unselected error, I found that adding the gain_path flag at the creating TS setting file step could clear the error.

alisterburt, 2025-01-15:
glad you got sorted!

closed 2025-01-15 by alisterburt

CleoShen, 2025-01-21:
I have a new tomostar unselected error when using ts_template_matching; the log as shown below, do you have any idea?
`2025-01-21 00:27:52.727 Received "TomoMatch", with 3 arguments, for GPU #0, 5856 MB free:
2025-01-21 00:27:52.949 Using 36864 orientations for matching
2025-01-21 00:27:53.019 0%
2025-01-21 00:27:53.110 Using 192 sub-volumes for matching, resulting in 338 % overhead
2025-01-21 00:27:53.180 0%
2025-01-21 00:27:53.249 Using 6, 6, 2 sub-volumes
2025-01-21 00:27:53.318 100%
2025-01-21 00:27:53.385 Loading...
2025-01-21 00:27:53.450 0%
2025-01-21 00:27:56.948 Preparing template...
2025-01-21 00:27:57.021 0%
2025-01-21 00:28:04.723 Matching...
2025-01-21 00:28:04.796 0%`

alisterburt, 2025-01-21:
@CleoShen you only had 6GB memory free at the start of template matching, this is likely a crash due to memory constraints. You could try reducing --batch_angles