warpem/warp #338: ts_etomo_fiducials Tilt Series Alignment - Wrong alignment of the fiducials

PierrePaillard, 2025-02-03:
Hi all, 

I'm running Warp2 on tilt-series that contain fiducials (10 nm diameter).

For the tilt-series alignment, I've run the command:
WarpTools ts_etomo_fiducials \
--settings warp_tiltseries.settings \
--angpix 10 \
--fiducial_size 10 \
(--n_beads_target 20)

After the Tilt Series: CTF Estimation > Reconstruct Tomograms commands run, the tomogram reconstruction is horrible. The fiducials produce a sort of dark lines in the whole thickness of the tomogram.

I already did several manual tomogram reconstructions by using fiducials on etomo (parameters, diameter of the fiducial: 10nm, nb of fiducials to generate as seed 20) and the alignment worked very well.
I also tried with ts_aretomo but output the same result as before...

Did it happens to someone else? 
Could somebody explain me what is the problem and what do I have to run differently for good fiducial alignment using Warp2?

I would like to use only Warp2 for the processing rather than using RELION for the first step and import the alignment on Warp.

Thank you!!

alisterburt, 2025-02-03:
Hi @PierrePaillard - thanks for taking the time to write up your issue!

I wrote both the RELION IMOD programs and the Warp IMOD programs and they're supposed to be ~identical.
I suspect your tilt series data may contain hot pixels which are not being removed and this is making automated tilt series alignment fail c.f. #231 and https://groups.google.com/u/1/g/warp-em/c/vuDInlQuN4A

To troubleshoot:
- can you rerun motion correction and check worker logs for a warning about switching to fp32? this would confirm my theory
- check the batchruntomo logs to see if they point at an obvious failure point
- open the .edf file in the IMOD directory for each tilt series (`etomo <ts_name>.edf` in the right directory or open from the GUI) to try to find where things went wrong with more visual feedback

You can always import alignments done outside of our wrapper programs following the docs here https://warpem.github.io/warp/reference/warptools/custom_tilt_series_alignments/

PierrePaillard, 2025-02-04:
Hi @alisterburt - thank you for your quick response!

I checked the batchruntomo log and the worker log after motion correction, but didn't find any warning about switching to fp32.

However after tomogram reconstruction, I found in the align.log file a warning message "the rotation angle seems to be closer to -94.4 than to the specified angle, -174.4". 
Notice that the angle in the .mdoc file is -174.4.

I re-run the tilt-serie alignment command, changing the axis to -94.4: 
WarpTools ts_etomo_fiducials --settings warp_tiltseries.settings --angpix 10 --initial_axis -94.4 --fiducial_size 10 --n_beads_target 20.

After reconstruction, the tomograms looked very well!
I run in parallel Relion5 pre-processing on the same dataset, and observed a difference in the alignment of the tilt-series. The alignment in relion looked better, as the fiducials display a better circle shape. 

Do you know why warp2 don't produces the same alignment as relion5 ? What should I change in my settings to improve the Warp alignment ?

Thank you!!

 
Here is an example of reconstructed tomogram I obtained before changing the axis in the tilt-alignment job. In case it happens to someone else ; ).
[image]

alisterburt, 2025-02-04:
@PierrePaillard tilt axis angles in mdoc files are often wrong e.g. here where the correct angle is in the header but `RotationAngle` entries are wrong

https://github.com/teamtomo/mdocfile/blob/20a7f87b19aa66be8b26c6585b0d33b056376605/tests/test_data/tilt_series.mdoc#L8-L21

Do you have the same numbers of fiducials in both your RELION IMOD run and your Warp IMOD run?

PierrePaillard, 2025-02-05:
@alisterburt Thank you for your response!!

Yes, I have the same number of fiducials in both RELION IMOD run and Warp IMOD run, since I used the same raw data set.
Also in the both cases, I've specified a defined number of fiducials to select for tilt-Alignment.

I don't know where the issue could originates...

alisterburt, 2025-02-05:
@PierrePaillard you need to dig a little deeper yourself, compare how the fiducials were tracked in each case and see where there are differences - if there are significant difference with ~the same inputs to the RELION and Warp programs then I'll look into it

alisterburt, 2025-04-28:
closed by #372 - extra warnings provided

closed 2025-04-28 by alisterburt