warpem/warp #339: warp frameseries dose weighting issue

LabBMH, 2025-02-04:
Hey,

I tested the Linux WARP frame series (for motion correction and CTF correction) on an internal test dataset (TIFF movies from a K3 detector). I ran all the data through the same cryoSPARC pipeline, using identical initial volumes and settings, and compared the reconstructions (normalized to 200,000 randomly selected particles) with those obtained from cryoSPARC-processed movies (using patch motion/patch CTF) and from Windows WARP–picked particles. Interestingly, the visual map quality and achieved resolution varied significantly between Linux WARP, cryoSPARC, and Windows WARP: Linux WARP produced by far the worst results, while cryoSPARC and Windows WARP yielded roughly comparable outcomes after global and local CTF refinements.

I believe this discrepancy is related to dose weighting. In the noise model, the cryoSPARC and Windows WARP–processed movies show an increase in noise variance at higher spatial frequencies, whereas the Linux WARP–processed movies display a flat noise variance. To my knowledge, this suggests a potential bug in the dose weighting algorithm for the Linux WARP version, where later frames are weighted equally to earlier frames.

I attached a PDF with the plots from the cryoSPARC processing.

[linux_warp_issues.pdf](https://github.com/user-attachments/files/18657512/linux_warp_issues.pdf)

Software used:
	•	Windows WARP: v1.09
	•	Linux WARP: v2.0.0 (running on Ubuntu 22.04 LTS)
	•	cryoSPARC: v4.6.2

Movies:
	•	Detector: K3 (Krios G3i)
	•	56 frames per movie
	•	0.89 e⁻/Å² per frame
	•	2488 micrographs

WARP processing parameters (identical for both Windows and Linux versions):
	•	CTF:
	•	Window: 512 px
	•	Range: 20–4 Å
	•	Voltage: 300 kV
	•	Cs: 2.7 mm
	•	Amplitude contrast: 0.07
	•	Defocus range: 0.2–4 µm
	•	Grid: 6×4×1
	•	Motion Correction:
	•	Considered resolution range: 40–4 Å
	•	B-factor: –500
	•	Grid: 6×4×56
	•	Binning:
	•	Bin 1, resulting in 0.5076 Å/px

alisterburt, 2025-02-04:
Hi @LabBMH - thanks for running this careful test and reporting the results, it's really useful! 

We haven't yet taken the time to fully validate the frame series pipeline - I'll discuss this with Dimitry and get back to you 🙂

LabBMH, 2025-02-04:
Hey, 

Thanks a lot 🙂!

alisterburt, 2025-02-05:
@LabBMH this is reasonably high on our priority list but no guarantees about timeline 🙂

LabBMH, 2025-02-05:
@alisterburt I made one more comparison, which might be interesting: I re-extracted the Windows warp-picked particles in cryoSPARC from the Windows warp averages (patch CTF corrected in cryoSPARC). This might be a more apples-to-apples comparison with the Linux warp fs_export_particles command (except for the patch CTF). It appears as if there are differences in how the movies are processed and the averages are written out. I am not sure why but maybe it might have to do with the dose weighting: For the Windows warp averages, you get a very high peak at lower spatial frequencies (around 5-6x higher) and a falling curve towards higher spatial frequencies. Even though the overall resolution is more similar to the Windows warp picked/exported particles, the b factor is similar to the Linux warp picked/exported particles, and also the map quality is much worse than the Windows picked/exported particles reconstruction.

I attached the results: 

[linux_warp_issues_part_2.pdf](https://github.com/user-attachments/files/18675904/linux_warp_issues_part_2.pdf)