warpem/warp #344: issue with fs_motion_and_ctf (error: Use the change_selection WarpTool to reactivate this item if required.)

ThatIsMyUsername, 2025-02-15:
Hi,
**When I run WarpTools fs_motion_and_ctf, the following message appears:**

```
No alternative input specified, will use input parameters from warp_frameseries.settings
File search will be relative to /data/zcy/Relion--tomo5.0/XMU-all/frames
1985 files found
Parsing previous results for each item, if available...
1985/1985, previous metadata found for 0                                                                                                                                                                                                                                                              
Connecting to workers...
Connected to 2 workers
Failed to process /data/zcy/mainwarp/xmu-all/warp_frameseries/Position_0018_011_-15.00_20231019_003756_fractions.mrc, marked as unselected                                                                                                                                                            
Check logs in /data/zcy/mainwarp/xmu-all/warp_frameseries/logs for more info.
Use the change_selection WarpTool to reactivate this item if required.
1985/1985, 1 failed, 00:00 remaining
```

**The log information is as follows:**

```
2025-02-14 13:53:04.462 Received "LoadStack", with 4 arguments, for GPU #0, 36550 MB free:
2025-02-14 13:53:08.329 Loaded stack: 11520, 8184, 10, normal, real, 1 A/px, ID = -1, 1
2025-02-14 13:53:08.329 Execution took 3.867 seconds
2025-02-14 13:53:08.330
2025-02-14 13:53:08.331 Received "MovieProcessMovement", with 2 arguments, for GPU #0, 33470 MB free:
2025-02-14 13:53:10.469 Processed movement for /data/zcy/mainwarp/xmu-all/warp_frameseries/Position_0018_011_-15.00_20231019_003756_fractions.mrc
2025-02-14 13:53:10.470 Execution took 2.138 seconds
2025-02-14 13:53:10.470
2025-02-14 13:53:10.472 Received "MovieExportMovie", with 2 arguments, for GPU #0, 31512 MB free:
2025-02-14 13:53:13.996 0: 1020
2025-02-14 13:53:13.996 1: 0
2025-02-14 13:53:13.997 2: 77
2025-02-14 13:53:13.997 3: 14
2025-02-14 13:53:13.997 4: 2559
2025-02-14 13:53:13.997 5: 3
2025-02-14 13:53:13.997 6: 0
2025-02-14 13:53:13.998 7: 26
2025-02-14 13:53:13.998 Exported movie for /data/zcy/mainwarp/xmu-all/warp_frameseries/Position_0018_011_-15.00_20231019_003756_fractions.mrc
2025-02-14 13:53:13.999 Execution took 3.526 seconds
2025-02-14 13:53:13.999
2025-02-14 13:53:14.001 Received "LoadStack", with 4 arguments, for GPU #0, 37076 MB free:
2025-02-14 13:53:14.238 Loaded stack: 11520, 8184, 1, normal, real, 1 A/px, ID = -1, 1
2025-02-14 13:53:14.238 Execution took 0.236 seconds
2025-02-14 13:53:14.238
2025-02-14 13:53:14.241 Received "MovieProcessCTF", with 2 arguments, for GPU #0, 37242 MB free:
2025-02-14 13:53:15.245 System. OverflowException: Value was either too large or too small for a Decimal.
2025-02-14 13:53:15.245    at System.Number. ThrowOverflowException(String message)
2025-02-14 13:53:15.245    at System.Decimal.DecCalc. VarDecFromR8(Double input, DecCalc& result)
2025-02-14 13:53:15.245    at Warp.Movie.<>c. <ProcessCTF>b__314_13(Double[] input, CTF ctf, Single[] defocusValues, Single[] phaseValues) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/Movie.cs:line 1710
2025-02-14 13:53:15.245    at Warp.Movie.<>c__DisplayClass314_4. <ProcessCTF>b__14(Double[] input) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/Movie.cs:line 1737
2025-02-14 13:53:15.245    at Accord.Math.Optimization.BaseOptimizationMethod.<>c__DisplayClass27_0. <Maximize>b__0(Double[] x)
2025-02-14 13:53:15.245    at Accord.Math.Optimization.LBFGS. line_search_morethuente(Int32 n, Double[] x, Double& f, Double[] g, Double[] s, Double& stp, Double[] xp, Double[] gp, Double[] wa, callback_data_t& cd, lbfgs_parameter_t& param)
2025-02-14 13:53:15.245    at Accord.Math.Optimization.LBFGS. lbfgs(Int32 n, Double[] x, Double& ptr_fx, lbfgs_evaluate_t proc_evaluate, lbfgs_progress_t proc_progress, Object instance, lbfgs_parameter_t _param)
2025-02-14 13:53:15.245    at Accord.Math.Optimization.BroydenFletcherGoldfarbShanno.Optimize()
2025-02-14 13:53:15.245    at Accord.Math.Optimization.BaseOptimizationMethod.Maximize()
2025-02-14 13:53:15.245    at Accord.Math.Optimization.BaseGradientOptimizationMethod.Maximize()
2025-02-14 13:53:15.245    at Warp.Movie. ProcessCTF(Image originalStack, ProcessingOptionsMovieCTF options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/Movie.cs:line 1902
2025-02-14 13:53:15.245    at WarpWorker.WarpWorker. EvaluateCommand(NamedSerializableObject Command) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpWorker/WarpWorker.cs:line 265

```

**I use the change_selection WarpTool to reactivate Position_0018_011_-15.00_20231019_003756_fractions.mrc：**
```
WarpTools change_selection \
--settings /data/zcy/mainwarp/xmu/warp_frameseries.settings \
--select \
--input_data /data/zcy/Relion--tomo5.0/XMU-all/frames/Position_0018_011_-15.00_20231019_003756_fractions.mrc
```

**Check the log again, the log information is correct, why?**

```
2025-02-14 14:14:40.978 Received "LoadStack", with 4 arguments, for GPU #0, 36502 MB free:
2025-02-14 14:14:45.138 Loaded stack: 11520, 8184, 10, normal, real, 1 A/px, ID = -1, 1
2025-02-14 14:14:45.139 Execution took 4.160 seconds
2025-02-14 14:14:45.139 
2025-02-14 14:14:45.141 Received "MovieProcessMovement", with 2 arguments, for GPU #0, 36502 MB free:
2025-02-14 14:14:46.356 Processed movement for /data/zcy/mainwarp/xmu-all/warp_frameseries/Position_0018_010_15.00_20231019_003638_fractions.mrc
2025-02-14 14:14:46.356 Execution took 1.214 seconds
2025-02-14 14:14:46.359 
2025-02-14 14:14:46.365 Received "MovieExportMovie", with 2 arguments, for GPU #0, 36862 MB free:
2025-02-14 14:14:49.941 0: 1064
2025-02-14 14:14:49.941 1: 0
2025-02-14 14:14:49.941 2: 73
2025-02-14 14:14:49.941 3: 16
2025-02-14 14:14:49.941 4: 2528
2025-02-14 14:14:49.942 5: 2
2025-02-14 14:14:49.942 6: 0
2025-02-14 14:14:49.942 7: 20
2025-02-14 14:14:49.942 Exported movie for /data/zcy/mainwarp/xmu-all/warp_frameseries/Position_0018_010_15.00_20231019_003638_fractions.mrc
2025-02-14 14:14:49.943 Execution took 3.577 seconds
2025-02-14 14:14:49.943 
2025-02-14 14:14:49.944 Received "LoadStack", with 4 arguments, for GPU #0, 36502 MB free:
2025-02-14 14:14:50.164 Loaded stack: 11520, 8184, 1, normal, real, 1 A/px, ID = -1, 1
2025-02-14 14:14:50.165 Execution took 0.220 seconds
2025-02-14 14:14:50.165 
2025-02-14 14:14:50.167 Received "MovieProcessCTF", with 2 arguments, for GPU #0, 36502 MB free:
2025-02-14 14:14:52.206 Processed CTF for /data/zcy/mainwarp/xmu-all/warp_frameseries/Position_0018_010_15.00_20231019_003638_fractions.mrc
2025-02-14 14:14:52.206 Execution took 2.039 seconds
2025-02-14 14:14:52.206 

```

thank you.

alisterburt, 2025-02-15:
@ThatIsMyUsername seems like you have hot pixels in your data that are causing issues, hot pixel removal was removed a while ago as it was hard to make robust with counting data - it looks like we might need to bring it back

Given that it's only one frame series, leaving it out should be fine 🙂

ThatIsMyUsername, 2025-02-16:
Thank you for your reply. Following the suggestion of [#296](https://github.com/warpem/warp/issues/296), I fixed the problem by changing the value of --c_range_max from 7 to 8. But why is that?

alisterburt, 2025-02-16:
Could you please send me a link to download the file?

ThatIsMyUsername, 2025-02-16:
Sorry, the dataset has not been made public yet. Do you need to view any information?

ThatIsMyUsername, 2025-02-18:
Hi @alisterburt , thank you very much for your reply. I successfully solved this problem by setting the appropriate parameters --c_range_max 5 and --c_range_min 50.
The correct setting of the c_range_max parameter is helpful for the successful operation of WarpTools ts_motion_and_ctf.
Thanks again.