warpem/warp #349: Error in WarpTools ts_reconstruct with the --halfmap_frames flag.

Kathir-66, 2025-02-28:
I am trying to obtain even and odd tomograms from frame series for performing denoising. 

When I run the command 

`WarpTools ts_reconstruct --settings frameseries.settings --angpix 13.056 --halfmap_frames `

I get the following error 

`Unhandled exception. System.InvalidCastException: Unable to cast object of type 'Warp.Movie' to type 'Warp.TiltSeries'.`

<detail>
<summary> Full log below </summary> 
`(warp) [kaezhila@cryog802 OPA_KO_v2]$ WarpTools ts_reconstruct --settings frameseries.settings --angpix 13.056 --halfmap_frames 
Running command ts_reconstruct with:
angpix = 13.056
halfmap_frames = True
halfmap_tilts = False
deconv = False
deconv_strength = 1
deconv_falloff = 1
deconv_highpass = 300
keep_full_voxels = False
dont_invert = False
dont_normalize = False
dont_mask = False
dont_overwrite = False
subvolume_size = 64
subvolume_padding = 3
device_list = {  }
perdevice = 1
workers = {  }
settings = frameseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null
input_norawdata = False

No alternative input specified, will use input parameters from frameseries.settings
File search will be relative to /sbdata/RAW/Krios1_K3/2023/2023-08-11_ledietri_tomo_OPA1_KO/frames
3138 files found
Parsing previous results for each item, if available...
3138/3138, previous metadata found for 3138                                                                                                                                                                                                                                             
Unhandled exception. System.InvalidCastException: Unable to cast object of type 'Warp.Movie' to type 'Warp.TiltSeries'.
   at WarpTools.Commands.ReconstructTiltseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Commands/Tiltseries/ReconstructTiltseries.cs:line 81
   at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted
`
</details>

Here Is the Complete list of steps I performed from the start
1. Creating the settings file
`(warp) [kaezhila@cryo101 OPA_KO_v2]$ WarpTools create_settings --folder_processing processing/ -o frameseries.settings --folder_data /sbdata/RAW/Krios1_K3/2023/2023-08-11_ledietri_tomo_OPA1_KO/frames/ --recursive --extension *.tif --angpix 1.088 --bin 1 --exposure -0.17897 --gain_path /sbdata/RAW/Krios1_K3/2023/2023-08-11_ledietri_tomo_OPA1_KO/frames/SuperRef_alignimages_000_0.0.dm4
`
2. fs_motion_and_ctf
`WarpTools fs_motion_and_ctf --settings frameseries.settings --m_range_min 200 --m_grid 5x5x12 --c_defocus_min 2.5 --c_defocus_max 5 --c_range_max 7 --out_averages --out_average_halves --device_list 0 1 2 3 
 `

Now when I try the ts_reconstruct I get the error I mentioned above!

Kathir-66, 2025-02-28:
Here Is the Complete list of steps I performed from the start
1. Creating the settings file
`(warp) [kaezhila@cryo101 OPA_KO_v2]$ WarpTools create_settings --folder_processing processing/ -o frameseries.settings --folder_data /sbdata/RAW/Krios1_K3/2023/2023-08-11_ledietri_tomo_OPA1_KO/frames/ --recursive --extension *.tif --angpix 1.088 --bin 1 --exposure -0.17897 --gain_path /sbdata/RAW/Krios1_K3/2023/2023-08-11_ledietri_tomo_OPA1_KO/frames/SuperRef_alignimages_000_0.0.dm4
`
2. fs_motion_and_ctf
`WarpTools fs_motion_and_ctf --settings frameseries.settings --m_range_min 200 --m_grid 5x5x12 --c_defocus_min 2.5 --c_defocus_max 5 --c_range_max 7 --out_averages --out_average_halves --device_list 0 1 2 3 
 `

Now when I try the ts_reconstruct I get the error I mentioned above!

alisterburt, 2025-02-28:
Hi @Kathir-66 - you are using a frame series settings file not a tilt series settings file... please see the tutorial for details 🙂

alisterburt, 2025-02-28:
Oh, and you are missing many steps before reconstructing a tomogram...

closed 2025-02-28 by alisterburt

Kathir-66, 2025-02-28:
Hi, sorry for my ignorance. This is the first time I am doing it, and thanks for getting back so fast. 

I looked into this tutorial for denoising https://warpem.github.io/warp/reference/warptools/tomogram_denoising/

We can generate even and odd movie averages by adding the --out_average_halves flag when running [WarpTools fs_motion_and_ctf](https://warpem.github.io/warp/reference/warptools/api/frame_series/#fs_motion_and_ctf). If you didn't do this don't worry, you can run [WarpTools fs_export_micrographs](https://warpem.github.io/warp/reference/warptools/api/frame_series/#fs_export_micrographs) with the --average_halves flag.

This will produce even and odd subfolders filled with images in the average folder of your frame series processing directory.
Generating even and odd tomograms

Now that we have even and odd images, producing even and odd tomograms is as simple as running [WarpTools ts_reconstruct](https://warpem.github.io/warp/reference/warptools/api/tilt_series/#ts_reconstruct) with the --halfmap_frames flag.

So this is the steps I followed.

alisterburt, 2025-02-28:
This page assumes you already have a complete warp processing directory containing the necessary metadata for performing reconstruction and is only telling you the specific steps required for tomogram denoising...

It sounds like you would benefit from completing the tutorial to understand how things are supposed to work
https://warpem.github.io/user_guide/warptools/quick_start_warptools_tilt_series/