warpem/warp #350: Update AreTomo wrapper to use AreTomo3?

shahpnmlab, 2025-03-02:
Just tested AreTomo3 commit [b58f844](https://github.com/czimaginginstitute/AreTomo3/commit/b58f84425082f578fc434cde6fc4fd2ea523f0e8) with `-OutImod 1`  flag and the porgram is now outputting sensible files that should work with WarpTools. The command i used was -
`AreTomo3 -InPrefix lam01_pos01_ts_003 -InSuffix .st  -OutDir ./  -VolZ 250 -AlignZ 400 -TiltAxis 84.7 -OutBin 2 -FlipVol 1  -PixSize 7.776 -OutImod 1 -Cmd 1  -Serial 1`
Note that you can no longer provide `-InMrc` and `-OutMrc` flags, instead one now needs to use a `-InPrefix` and `-InSuffix` flags. 
Output in .tlt file
```
-54.00
  -51.00
  -48.00
  -45.00
  -42.01
  -39.01
  -36.00
  -33.01
  -30.01
  -27.01
  -24.00
  -21.01
  -18.01
  -15.01
  -12.00
   -9.00
   -6.01
   -3.01
   -0.01
    2.99
    5.98
    8.99
   11.99
   14.99
   17.99
   20.99
   23.98
   26.99
   29.99
   32.99
   35.98
   38.98
   41.98
   44.98
   47.98
   50.99
   53.98
   56.98
   59.97
```
Output in .xf file
```
0.097     0.995    -0.995     0.097     -4.06      0.42
    0.097     0.995    -0.995     0.097    -14.28     -2.20
    0.097     0.995    -0.995     0.097    -17.16      2.16
    0.097     0.995    -0.995     0.097    -10.85     -0.47
    0.097     0.995    -0.995     0.097     -7.82     -1.99
    0.097     0.995    -0.995     0.097     -2.27      1.78
    0.097     0.995    -0.995     0.097     -4.21     -0.58
    0.097     0.995    -0.995     0.097    -12.53     -1.64
    0.097     0.995    -0.995     0.097     -3.27      1.75
    0.097     0.995    -0.995     0.097     -2.41      0.03
    0.097     0.995    -0.995     0.097     -7.18     -1.09
    0.097     0.995    -0.995     0.097     -0.54      1.79
    0.097     0.995    -0.995     0.097     -2.44     -0.34
    0.097     0.995    -0.995     0.097     -5.09     -0.98
    0.097     0.995    -0.995     0.097      2.42      1.24
    0.097     0.995    -0.995     0.097      0.96      0.29
    0.097     0.995    -0.995     0.097     -1.85     -0.90
    0.097     0.995    -0.995     0.097     -0.68     -0.91
    0.097     0.995    -0.995     0.097     -0.00      0.00
    0.097     0.995    -0.995     0.097      0.66     -1.04
    0.097     0.995    -0.995     0.097      7.90     -1.20
    0.097     0.995    -0.995     0.097     -3.47      0.03
    0.097     0.995    -0.995     0.097      1.46      0.65
    0.097     0.995    -0.995     0.097     -3.58     -1.78
    0.097     0.995    -0.995     0.097     -0.49      0.30
    0.097     0.995    -0.995     0.097      4.51      1.70
    0.097     0.995    -0.995     0.097      1.40     -1.37
    0.097     0.995    -0.995     0.097     -7.13     -1.87
    0.097     0.995    -0.995     0.097     -0.09      0.34
    0.097     0.995    -0.995     0.097      3.49     -2.46
    0.097     0.995    -0.995     0.097      7.94      0.64
    0.097     0.995    -0.995     0.097     -1.40     -0.40
    0.097     0.995    -0.995     0.097     -2.24     -2.57
    0.097     0.995    -0.995     0.097     -2.58     -1.33
    0.097     0.995    -0.995     0.097    -18.15     -0.35
    0.097     0.995    -0.995     0.097    -11.31     -4.93
    0.097     0.995    -0.995     0.097     -5.98     -1.57
    0.097     0.995    -0.995     0.097    -36.03     -2.03
    0.097     0.995    -0.995     0.097    -24.02     -4.23
```
output in newst.com
```
cat newst.com
$newstack -StandardInput
InputFile	lam01_pos01_ts_003.mrc
OutputFile	lam01_pos01_ts_003_st.ali
TransformFile	lam01_pos01_ts_003_st.xf
TaperAtFill     1,0
AdjustOrigin
OffsetsInXandY  0.0,0.0
#DistortionField        .idf
ImagesAreBinned 1.0
BinByFactor     1
#GradientFile   hc20211206_804.maggrad
```
Hope this helps.

shahpnmlab, 2025-04-28:
There is no perceived benefit to including aretomo3 wrappers... closing for now...

closed 2025-04-28 by shahpnmlab

alisterburt, 2025-04-28:
Actually the auto thickness is interesting...

alisterburt, 2025-09-24:
thanks again to @davidetorre99 !