warpem/warp #355: Anaconda defaults referenced through pytorch channel

stephen-riggs, 2025-03-14:
Hi Alister,

We've been removing programs which reference the paid-for "defaults" anaconda channels and warp was flagged up in the process of that.

As you've noted in issue #206 , you don't actually install any packages from the defaults channel, but you do use the pytorch channel which adds defaults to the channels list.
This happens in https://github.com/warpem/warp/actions/runs/13687663586/job/38274631291#step:5:13 

Pytorch themselves now say to install using pip, but the specific version you have in your dependencies is only available through conda.

Feel free to close this as a non-issue if you're happy with this as it is, but just thought I'd flag it up.
From my side I'm going to build from source as we've been told we can't use anything which references the defaults channel in any way.

alisterburt, 2025-03-14:
@stephen-riggs thanks for flagging this, I wasn't aware the pytorch channel added the default channel to the list and that's good to be aware of

FWIW we initially attempted to use pytorch from conda-forge (c.f. https://github.com/conda-forge/staged-recipes/pull/25492#issuecomment-2374628320) but ran into permissions errors with that package at the time

alisterburt, 2025-08-29:
@PertuyF I am very pro the conda-forge ecosystem but sorting out the pytorch build was too high complexity to be worth it to us. I'm not sure what benefit you perceive there is in publishing under another channel? If it's problems with anaconda inc, we are not using their packages so this is a non-issue.

PertuyF, 2025-09-01:
Thank you for the warning, I'll have a look and judge based on opportunity cost :)

closed 2025-09-24 by alisterburt