warpem/warp #360: Subtomogram export format

alncat, 2025-03-31:
Dear warp/m developers，
   Thank you very much for developing such great tools! I was wondering if warp can export subtomograms without ctf correction, while exporting per-particle CTF parameters as csv or star files for users.  I found that format might be beneficial for training neural networks for heterogeneity analysis (I am working on a project https://github.com/alncat/opusTomo). I have tested using a self modified warp and find the subtomograms without ctf correction works better.  In addition, can you help me clarify the difference between the subtomograms or image tilt series exported by M on windows, and the subtomograms exproted by Warp on linux after M refinement?  I noticed that M will update ***.xml files for micrographs after M refinement, which may update the image warp model and CTF model for micrographs.  You may point out the corresponding source code for me.

Sincerely,
Zhenwei

alisterburt, 2025-03-31:
Hey @alncat ! Thanks for reaching out - this isn't something that we currently support, if you submit a PR which adds the feature as optional functionality for particle export I would be open to adding it

The difference between particle export in M desktop and WarpTools is that temporally resolved particle poses are not taken into account in WarpTools

alncat, 2025-04-01:
Hi @alisterburt , thank you very much for the answer! I will dig deep into code and make the feature as optional functionality~

alisterburt, 2025-04-01:
@alncat I thought you already had it working? 

Your statement about getting worse results with CTF corrected subtomograms implies that maybe you're interpreting the data incorrectly in your model - are you dealing with CTF on these volumes?

This is not a feature that we particularly want or need ourselves and you might be better served by trying to work out how to use the volumes we already produce... uncorrected volumes should not work better

alncat, 2025-04-01:
@alisterburt The underlying training process of my program is that the neural network will reconstruct raw volume, then the raw volume will be multiplied with CTF (reconstructed using a pytorch script on the fly), the ctf corrected volume will then match with the subtomogram. In principle, the ctf corrected subtomogram should also work.

I did a hack by commenting the ctf correction part in reconstruct subtomogram. WARP can then output the subtomogram without ctf correction. Then i trained 3DCNN with this volume and the ctf parameters (CTF volume is reconstructed by my program, which simply backprojects CTF slice to different angles and sums them up). This approach seems to be better than using the ctf corrected subtomogram, and ctf volume output by WARP. The later approach might requires different initialization for neural networks since I found out that ctf volume is standardized.

Another way i have tried is simply using the image tilt series exported by M. I then manually reconstructed subtomogram using Relion (I found out that the WARP tools using the similar reconstruction algorithm as the relion lib), so this approach should be equivalent to reconstruct subtomogram without ctf correction.  Training with the subtomogram reconstructed by Relion without gridding, and CTF reconstructed by my program also seems working (I have documented the process here https://github.com/alncat/opusTomo/wiki/Convert-M's-2D-particle-image-series-to-subtomograms-for-training-OPUS%E2%80%90TOMO).

alncat, 2025-04-01:
Hey @alisterburt , will the linux version of M support exporting the particle tilt series like windows M ? That will be great!

alncat, 2025-04-02:
@alisterburt , I have also encountered a seemingly odd behavior when using WARP/M.  I found that it is necessary to do a round of subtomogram averaging before importing particles into M.  
For the same set of particle, if I import them directly into M, the resolution of the particle set will be infinity. However, if I did one round of subtomogram averaging (one iteration is enough) using pyTom, then importing the same set of particle can reach reasonable resolution.
I checked the particles.star files in corresponding folders. 

[image]

In the above image, the particle set on the left is imported directly without refinement, the particle set on the right is imported after one iteration of refinement in pyTom.  The coordinates and euler angles are pretty close.  
However, the FSCs are different.

[image]

I was wondering if there is any explanations for this behavior.