warpem/warp #363: Warp conda installation issue

mapaulson, 2025-04-07:
Hi,
I'm trying to install the latest version of Warp on Linux using the conda installation outlined on the webpage. Unfortunately the install stalls while running the solver:

$ conda create -n warp warp -c warpem -c nvidia/label/cuda-11.8.0 -c pytorch -c conda-forge
Collecting package metadata (current_repodata.json): | WARNING conda.models.version:get_matcher(542): Using .* with relational operator is superfluous and deprecated and will be removed in a future version of conda. Your spec was 1.7.1.*, but conda is ignoring the .* and treating it as 1.7.1
done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): - 

and remains at this point indefinitely (i.e. at least a day). I'm using standard Ubuntu 22.04 Linux and have had no issues with any other conda installation. Any advice appreciated!

alisterburt, 2025-04-07:
Hi @mapaulson - I just tried and can't reproduce your issue
<details><summary>install log</summary>

```txt
conda create -n warptest warp=2.0.0 -c warpem -c nvidia/label/cuda-11.8.0 -c pytorch -c conda-forge
pytorch/linux-64                                              No change
pytorch/noarch                                      10.2kB @  86.1kB/s  0.0s
warpem/linux-64                                               No change
nvidia/label/cuda-11.8.0/noarch                               No change
warpem/noarch                                                 No change
nvidia/label/cuda-11.8.0/linux-64                             No change
nodefaults/linux-64                                125.0 B @ 431.0 B/s  0.2s
nodefaults/noarch                                  116.0 B @ 351.0 B/s  0.2s
conda-forge/noarch                                  19.8MB @  40.1MB/s  0.3s
conda-forge/linux-64                                43.1MB @  56.7MB/s  0.6s

Transaction

  Prefix: /redacte/path/burta2/micromamba/envs/warptest

  Updating specs:

   - warp=2.0.0

  Package                          Version  Build                         Channel                        Size
───────────────────────────────────────────────────────────────────────────────────────────────────────────────
  Install:
───────────────────────────────────────────────────────────────────────────────────────────────────────────────

  + cuda-documentation             11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-nvml-dev                  11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-cuobjdump                 11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-cuxxfilt                  11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-nvcc                      11.8.89  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-nvprune                   11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-cudart                    11.8.89  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-nvrtc                     11.8.89  0                             nvidia/label/cuda-11.8.0     Cached
  + libcublas                    11.11.3.6  0                             nvidia/label/cuda-11.8.0     Cached
  + libcufft                     10.9.0.58  0                             nvidia/label/cuda-11.8.0     Cached
  + libcufile                     1.4.0.31  0                             nvidia/label/cuda-11.8.0     Cached
  + libcurand                    10.3.0.86  0                             nvidia/label/cuda-11.8.0     Cached
  + libcusolver                  11.4.1.48  0                             nvidia/label/cuda-11.8.0     Cached
  + libcusparse                  11.7.5.86  0                             nvidia/label/cuda-11.8.0     Cached
  + libnpp                       11.8.0.86  0                             nvidia/label/cuda-11.8.0     Cached
  + libnvjpeg                    11.9.0.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-cccl                      11.8.89  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-driver-dev                11.8.89  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-profiler-api              11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-cupti                     11.8.87  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-gdb                       11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-memcheck                  11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-nvdisasm                  11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-nvprof                    11.8.87  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-nvtx                      11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-sanitizer-api             11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-nsight                    11.8.86  0                             nvidia/label/cuda-11.8.0     Cached
  + nsight-compute             2022.3.0.22  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-compiler                   11.8.0  0                             nvidia/label/cuda-11.8.0     Cached
  + cuda-nvrtc-dev                 11.8.89  0                             nvidia/label/cuda-11.8.0     Cached
  + libcublas-dev                11.11.3.6  0                             nvidia/label/cuda-11.8.0     Cached
  + libcufft-dev                 10.9.0.58  0                             nvidia/label/cuda-11.8.0     Cached
  + gds-tools                     1.4.0.31  0                             nvidia/label/cuda-11.8.0     Cached
  + libcufile-dev                 1.4.0.31  0                             nvidia/label/cuda-11.8.0     Cached
  + libcurand-dev                10.3.0.86  0                             nvidia/label/cuda-11.8.0     Cached
  + libcusolver-dev              11.4.1.48  0                             nvidia/label/cuda-11.8.0     Cached
  + libcusparse-dev              11.7.5.86  0                             nvidia/label/cuda-11.8.0     Cached
[... cut ...]
Linking pytorch-2.0.1-py3.11_cuda11.8_cudnn8.7.0_0
Linking torchtriton-2.0.0-py311
Linking torchvision-0.15.2-py311_cu118
Linking warp-2.0.0dev33-py311_0

Transaction finished

To activate this environment, use:

    micromamba activate warptest

Or to execute a single command in this environment, use:

    micromamba run -n warptest mycommand

</details>

is it possible you installed a bunch of things into your base environment?
c.f. https://github.com/mamba-org/mamba/issues/2208#issuecomment-1374712154

mapaulson, 2025-04-08:
Ah, looks like my bad. I was using the Anaconda installer, not Miniforge. Switching to Minforge solved the problem - I guess this relates to the Anaconda licensing changes.

closed 2025-04-08 by mapaulson