warpem/warp #364: WarpTools ts_export_particles --3d not exporting full gains from MCore refinement

deadlift385, 2025-04-07:
Hi Alister,

I did a few more tests regarding WarpTools ts_export_particles --3d. In panels A-F, the red FSC curve is after MCore refinement and the blue curve is after --3d particle export and relion_reconstruct. I calcualted all FSCs from the corresponding half maps from the two programs and by using the same mask for all calcuations.

Panel A, refinement iteration 0: MCore --iter 0

Panel B, refinement iteration 1: MCore --refine_particles

Panel C, refinement iteration 2: MCore --refine_particles --refine_imagewarp 4x4

Panel D, refinement iteration 3: MCore --refine_particles --refine_imagewarp 4x4 --refine_stageangles

Panel E, refinement iteration 4: MCore --refine_particles --refine_imagewarp 4x4 --refine_stageangles --refine_volumewarp 4x4x2x10

Panel F, refinement iteration 5: MCore --refine_particles --refine_imagewarp 4x4 --refine_stageangles --refine_volumewarp 4x4x2x10 --ctf_defocus

As you can see, intially, without MCore refining anything or by just refining the particle poses, one obtains the same quality reconstructions after --3d particle export and relion_reconstruct (panels A and B). But then, after refinment of imagewarp etc., --3d particle export fails to export the full extent of the gain from the MCore refinements (panel C). The blue curve does get better, e.g. going from panel B to C, but it just lags more and more behind the red MCore curve.

This is all without weighting or temproal poses.

I did the same analysis with the apoferritin tutorial dataset and the results look similar, albeit less pronounced (probably because the apoferritin data has less distortions comparted to the virus dataset shown here).

So, why is it that WarpTools ts_export_particles --3d initially exports 3D particles that yield reconstuctions of the same quality, but with further MCore refinment the corrections appear not to be exported fully?

Best, Simon

[image]

alisterburt, 2025-04-07:
@deadlift385 interesting, thanks for sharing!

Your tests clearly show that particle export for RELION is not including many parts of the M model...

C is puzzling and I don't understand it. I looked at [`TiltSeries.ReconstructSubtomos.cs`](https://github.com/warpem/warp/blob/main/WarpLib/TiltSeries/TiltSeries.ReconstructSubtomos.cs) and the positions that are used for particle image extraction do take into account image warp from `GridMovementX/Y`, the image warping grids

https://github.com/warpem/warp/blob/215a235eac622b24118064b3539dc0731fb0dde2/WarpLib/TiltSeries/TiltSeries.cs#L555-L556

From what I can see these are the same image warping grids which are optimized in [`TiltSeries.MultiParticleRefinement.cs`](https://github.com/warpem/warp/blob/main/WarpLib/TiltSeries/TiltSeries.MPARefinement.cs)...

cc @dtegunov ?

deadlift385, 2025-04-08:
When going from B (#01) to C (#02) it clearly does export image warp, but not to the full extent.

Can we do cc @dtegunov ?

[image]

alncat, 2025-04-11:
I have refined 70S ribosome before, by refining subtomograms in M and reexporting subtomograms in WARP,  the STA in pyTOM can produce a subtomogram average with a resolution of 8A 
before M refinement
[image]
after M refinement
[image]

deadlift385, 2025-04-11:
It's not relion_reconstruct (see panels A and B).

shahpnmlab, 2025-04-14:
I have also experienced a similar pathology, the way I explained it away, was by considering the fact that M does not generate a *_unfil reconstruction at the end of a refinment run, instead it produces only filtered outputs (with half maps) for the user...

shahpnmlab, 2025-04-14:
Hi Simon,
Yes, thats what I mean.. the only unfiltered outputs are the two half-maps...