warpem/warp #365: Error during MCore refinement with --refine_tiltmovies

nkjaerjensen, 2025-04-14:
Hi Alister/Dimitry,

After initial MCore refinements running perfectly fine, I keep running into this error if I attempt to use --refine_tiltmovies in the following MCore run. I have inspected the metadata and cannot seem to find problems there. I am able to keep running further MCore refinements as long as I don't use --refine_tiltmovies. Reverting to the initial versions of the metadata (before any M runs) didn't help either and results in the same error. Only running --refine_tiltmovies without other refinements also results in the same error. Re-running all the initial MCore refinements in a new M population with new sources/species and .xmls that have been reset also reproduced the error when --refine_tiltmovies is added again. 

Any new ideas to get --refine_tiltmovies working would be much appreciated, I am happy to try any suggestions!

This happened in dev31.

With export WARP_DEBUG=1:

Error from terminal for first attempt with --refine_tiltmovies:
```
  what():  cuFFT error: CUFFT_INTERNAL_ERROR at /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/NativeAcceleration/gtom/src/FFT/FFT.cu:23

Unhandled exception. System.Net.Http.HttpRequestException: An error occurred while sending the request.
 ---> System.Net.Http.HttpIOException: The response ended prematurely. (ResponseEnded)
   at System.Net.Http.HttpConnection.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at System.Net.Http.HttpConnection.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.SendWithVersionDetectionAndRetryAsync(HttpRequestMessage request, Boolean async, Boolean doRequestAuth, CancellationToken cancellationToken)
   at System.Net.Http.RedirectHandler.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
   at Warp.WorkerWrapper.SendCommand(NamedSerializableObject command) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/WorkerWrapper.cs:line 228
   at Warp.WorkerWrapper.MPARefine(String path, String workingDirectory, ProcessingOptionsMPARefine options, DataSource source) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/WorkerWrapper.cs:line 458
   at MCore.MCore.<>c__DisplayClass16_4.<DoProcessing>b__4(Int32 ifile, Int32 threadID) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/MCore/MCore.cs:line 473
   at Warp.Tools.Helper.<>c__DisplayClass51_0.<ForCPU>b__0(Object id) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/Tools/Helper.cs:line 719
Aborted (core dumped)
```

All subsequent attempts with --refine_tiltmovies results in the same error as above or at other times, the attempt results in this error in the terminal:
```
MCore --population data.population --refine_imagewarp 4x4 --refine_particles --ctf_defocus --refine_tiltmovies
Loading population... Done
Creating directories... Done
Spawning workers... Done
Preparing for refinement – this will take a few minutes per species
Preparing refinement requisites...
$<5>                                                                                                                                                              $<5>1/1
Performing refinement
Preparing population for data source data1...Done
Loading gain reference for data1... Done
Refining all series in data source...
0/6terminate called after throwing an instance of 'std::runtime_error'
  what():  cuFFT error: CUFFT_INTERNAL_ERROR at /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/NativeAcceleration/gtom/src/FFT/FFT.cu:23

Unhandled exception. Unhandled exception. System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/MCore/MCore.cs:line 598
   at Warp.WorkerWrapper.ReportDeath() in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/WorkerWrapper.cs:line 255
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0() in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/WorkerWrapper.cs:line 189
Aborted (core dumped)
```

Error from worker_crash.txt is always:
```
System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at Warp.TiltSeries.<>c__DisplayClass101_1.<PerformMultiParticleRefinementOneTiltMovie>b__2(Boolean CorrectBeamTilt) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/TiltSeries.cs:line 8501
   at Warp.TiltSeries.PerformMultiParticleRefinementOneTiltMovie(String workingDirectory, ProcessingOptionsMPARefine optionsMPA, Species[] allSpecies, DataSource dataSource, Movie tiltMovie, Image[] tiltMovieData, Int32 tiltID, Dictionary`2 SpeciesParticles, Dictionary`2 SpeciesParticleSubsets, Dictionary`2 SpeciesParticleIDRanges, Dictionary`2 SpeciesContainmentMasks, Dictionary`2 SpeciesRefinementSize, Dictionary`2 SpeciesRelevantRefinementSizes, Dictionary`2 SpeciesFrameWeights, Dictionary`2 SpeciesCTFSuperresFactor) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/TiltSeries.cs:line 8772
   at Warp.TiltSeries.PerformMultiParticleRefinement(String workingDirectory, ProcessingOptionsMPARefine optionsMPA, Species[] allSpecies, DataSource dataSource, Image gainRef, DefectModel defectMap, Action`1 progressCallback) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpLib/TiltSeries.cs:line 5846
   at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1732560643671/work/WarpWorker/WarpWorker.cs:line 731
```

alisterburt, 2025-04-14:
thanks @nkjaerjensen for the super complete report - I'm actually not sure I've tried to use this flag myself since M moved to linux... 🫠 

I've let Dimitry know, I'm wondering if he has used it recently or whether this is somehow specific to your data

alisterburt, 2025-04-14:
Dimitry confirms he doesn't think he's ever tested it either... 😂

nkjaerjensen, 2025-04-15:
Thanks so much for the prompt reply @alisterburt !! Hopefully it is not a difficult fix.

nkjaerjensen, 2025-04-28:
Hi @alisterburt , just to update I've attempted to run the --refine_tiltmovies flag on other datasets of mine now as well and also attempted to make some changes to Tiltseries.cs and build dev31 myself to fix the IndexOutOfRangeException. but with no luck so far in fixing the error.

Importantly, I just tested and can confirm that the --refine_tiltmovies flag seems to work on the tutorial dataset (native dev31, run on your pre-calculated results https://zenodo.org/records/11398168). It also writes out averages etc. as expected. 

My /frames folder uses relative softlinks that point to the raw .eer data. Replacing the softlinks with the identically named actual raw .eer data enables M to run with --refine_tiltmovies, except no refinements are actually performed. Here is an M log from a tilt series refinement:

```
2025-04-28 15:29:46.608 Received "MPARefine", with 4 arguments, for GPU #1, 30562 MB free:
2025-04-28 15:29:46.737 Finished refining TS_005.tomostar
2025-04-28 15:29:46.740 Execution took 0.129 seconds
2025-04-28 15:29:46.742 
```

I am guessing some metadata somewhere is not happy, but after looking through the .xmls etc. I struggle to find where.

The main difference between the tutorial dataset and my data is that my raw data is .eer instead of .tif. Since all other functionality in warp/M seems to work for the way I have processed my data, I suspect it may have something to do with this data format and how it is handled? Let me know what you think!

nkjaerjensen, 2025-05-11:
After playing with this some more, my conclusion is that --refine_tiltmovies only works for .tif files and not if the frame series are .eer.

Converting my .eer data to .tif, reprocessing fs_motion_and_ctf, making changes in the metadata file paths to reflect the file format change and only using paths to the raw .tif frame series in the metadata (and never using paths to softlinks instead) successfully allows the use of the --refine_tiltmovies flag on my data. However, processing as .eer data without using softlinks (i.e. exactly as I process the .tif data that successfully uses the --refine_tiltmovies flag), I get the following error when I use --refine_tiltmovies:

 In worker0.err:
```
terminate called recursively
terminate called recursively
terminate called after throwing an instance of 'std::runtime_error'
terminate called recursively
terminate called recursively
terminate called recursively
  what():  Failed to open file
```

All other MCore refinements I have attempted work as usual on .eer data, also after doing as described above.

Closing for now.

alisterburt, 2025-05-12:
Thanks @nkjaerjensen for your efforts digging around here - it seems pretty conclusive and will make fixing the issue much easier, reopening until we can fix 🙂