warpem/warp #369: Poor quality of Warp tomogram reconstruction with imported etomo alignments

mpm896, 2025-04-24:
Hi all,

I'm trying to preprocess and reconstruct tomograms using `WarpTools` and I already have good alignments from Etomo. In Etomo, I aligned tomograms with a pixel size of 7.904 (i.e. bin 4). To import the alignments into `Warp` I ran `ts_import_alignments --settings warp_tiltseries.settings --alignments /path/to/alignments --alignment_angpix 7.904`
(where `/path/to/alignments` contains subdirectories for each dataset and **only** contains the `.xf` and `.tlt` files from the etomo alignment). I then ran (after `ts_ctf`): ` WarpTools ts_reconstruct --settings warp_tiltseries.settings --angpix 10 --deconv`

Here is a screenshot from the IMOD reconstructed tomogram (R-weighted back projection):

[image]

Here is a screenshot from the Warp reconstructed tomogram with imported etomo alignments:

[image]

Someone else told me that they found for some datasets, if they delete the `.xml` file inside `warp_tiltseries` directory, they get better reconstructions. I tried that, and you can see in the screenshot that features from the IMOD reconstruction are more visible (i.e. ice on the left, hole edge in bottom right) but it still isn't very good and the filaments aren't readily visible:

[image]

I also found that the Y and Z axes are flipped in the Warp reconstruction. 

Let me know if you've experienced this before if maybe there's some settings that I didn't properly set, or if I can provide any other info. Thanks!

mpm896, 2025-04-25:
Update: I ran

 `WarpTools ts_etomo_patches --settings warp_tiltseries.settings --angpix 7.904 --patch_size 1500 --initial_axis -3.00` 

(7.904 A/pix = bin 4, patch size of 1500 corresponds to the 200 pixels I typically set in Etomo). Reconstruction now looks great. However when I import good alignment from an external Etomo run, the reconstruction quality is still very poor

alisterburt, 2025-04-25:
Hi @mpm896 - sorry for the delay, have been travelling

The flip relative to external processing can be expected depending on the way tilt angles are handled, ideally you would run ts_stack then your own IMOD processing then import_alignments to ensure consistency... I thought the flip in tilt angles wouldn't matter for the xf but maybe it does somehow?

In any case I'm glad you're having auccess running the wrapper, it's not doing anything magic and the only thing I can think of is the tilt angles... maybe pixel size is incorrect somewhere?

alisterburt, 2025-04-25:
I don't expect the import has broken... it is what is used internally by ts_etomo/AreTomo jobs

mpm896, 2025-04-25:
Hey @alisterburt , no worries! Hope ccpem has been fun.

I've been comparing the XML files in the `warp_tiltseries` dir after importing vs running `ts_etomo_patches` and found pretty significant differences in the `AxisOffsetX` and `AxisOffsetY` sections (as well as small differences in `Angles` and `AxisAngle` that don't seem too significant). 

The initial Etomo runs I produced a bin 4 tilt series starting at coarse alignment, so I assumed everything was done at bin 4 dimensions and so I ran `ts_import` with `--alignment_angpix 7.904`. When doing this, `AxisOffsetX` and `AxisOffsetY` are way off compared to the XML from `ts_etomo_patches`. If I import the alignments with the unbinned pixel size (1.976) then `AxisOffsetX` is a bit closer and `AxisOffsetY` is closer but has the opposite sign.

Just to be clear, to import alignments I just have to provide the `.xf` and `.tlt` files?

I've attached the XML files just in case you want to take a look and may notice anything else off, because I don't see why anything else would be different

[fromTsEtomoPatches.txt](https://github.com/user-attachments/files/19912783/fromTsEtomoPatches.txt)
[fromBin4Import.txt](https://github.com/user-attachments/files/19912773/fromBin4Import.txt)
[fromUnbinnedImport.txt](https://github.com/user-attachments/files/19912771/fromUnbinnedImport.txt)

mpm896, 2025-04-25:
@alisterburt Found it! I have no idea why, but the 6th column of the `xf` file (DY) from an external Etomo run has the opposite sign as DY from the `xf` file produced when doing the alignment through `ts_etomo_patches`. I modified the external `xf` file with:

`awk '{ $6 = -$6; print }' file.xf.bak > file.xf`

then re-imported and reconstructed the tomogram, and bam, problem solved. However I don't understand why DY would be flipped between the two? Does Warp have a different convention for, say, how it orients the frames/tiltseries?

alisterburt, 2025-04-25:
Woah - super weird, great job spelunking!

No idea, how was the TS stack made? Any chance there is some extra IMOD metadata in there that leads to the flip? We probably don't deal with that if there is something

mpm896, 2025-04-25:
Ah, just checked the original, unaligned MRC file and the .st file produced by `ts_stack` - they are flipped over the Y axis

mpm896, 2025-04-25:
One last question - Have you used Warp reconstructed tomograms in Dynamo? Dynamo is refusing to read the volume because the map mode of the tomograms is 12 (16-bit float). Running `newstack -mode 1 in.mrc out.mrc` to convert to map mode 1 (16 bit int) produces a tomogram with essentially all pixel values as 0. I'm wondering if you have a workaround for this? Thanks!

mpm896, 2025-04-25:
Figured it out - had to rescale the pixel intensities before changing the map mode. Closing this issue. Thanks!

mpm896, 2025-08-20:
It seems like there is something fishy going on. I ran `ts_etomo_patches` to compare, and the reconstructions look the same. Rather than 2 halves of the fiducials coming together as you slice through, the fiducials take on a V shape with some smearing in the vertical-ish idrection.

I went into the etomo reconstruction directory made by `ts_etomo_patches` and it's really quite odd. Fiducial tracking looks great, but then computing the alignment gives insanely high errors and the resulting tomogram from etomo looks similar to the Warp reconstruction

mpm896, 2025-08-20:
[Spoke with David](https://groups.google.com/a/colorado.edu/g/imod/c/xlrxSgTRtMc/m/8_gplx90AAAJ) and for a flip in Y, he suggested the following modifications to the `xf` file:

```
echo '1 0 0 -1 0 0' > flipy.xf

xfproduct flipy.xf setname.xf temp.xf
xfproduct temp.xf flipy.xf setname.xf
```

This properly modifes the transforms to match the warp stack. Closing this issue again!

closed 2025-08-20 by mpm896