warpem/warp #370: Issue Using M on One Dataset with Two Users

jacob-r-anderson, 2025-04-26:
I have a use case in which two users (User_01 & User_02 (me)) are working on reconstructing different molecules from the same dataset. The data was processed in Warp by User_01 in their directory. The poses problems when at the MTools stage as I am unable to run the MTools create_source command without getting errors. I believe this is because MTools create_source is trying to write in the original processing directory of User #1.

```
User_02$ MTools create_source --population Pop_01/Pop_01.population --name 20240415 --processing_settings SettingsFiles/warp_tiltseries_20240415.settings
Running command create_source with:
population = Pop_01/Pop_01.population
processing_settings = SettingsFiles/warp_tiltseries_20240415.settings
name = 20240415
nframes = 
files = 

Looking for data... found 22 files
Loading metadata... 22/22                                                                                                                

Adding 22 items.
Calculating data hashes... 22/22                                                                                                         
Committing initial version... Unhandled exception. System.UnauthorizedAccessException: Access to the path 'User01/WarpM/warp_tiltseries_20240415/versions/91nsMIeE545V3kHKz1wtFDs1zgA/' is denied.
```

I tried to work around this problem by changing "ProcessingFolder" in the warp_tiltseries_20240415.settings file. For example I replaced:

`<Param Name="ProcessingFolder" Value="User01/WarpM/warp_tiltseries_20240415" />
`
with

`<Param Name="ProcessingFolder" Value="User02/WarpM/warp_tiltseries_20240415/" />
`

However, this has an error of its own.

```
User02$ MTools create_source --population Pop_01/Pop_01.population --name 20240415 --processing_settings SettingsFilesChanged/warp_tiltseries_20240415.settings
Running command create_source with:
population = Pop_01/Pop_01.population
processing_settings = SettingsFilesChanged/warp_tiltseries_20240415.settings
name = 20240415
nframes = 
files = 

Looking for data... found 22 files
Loading metadata... 22/22                                                                                                                

Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
```

I was curious if there is any work around to this problem. I think it might eventually be a general problem where multiple laboratory members want to work on the same dataset without needing write access to where the original data is processed or stored.

jacob-r-anderson, 2025-04-26:
For the moment I have been able to work around this by copying only the XML files from `User01/WarpM/warp_tiltseries_20240415` to `User02/WarpM/warp_tiltseries_20240415/`.

alisterburt, 2025-09-24:
closing - please feel free to reopen if you'd like to continue discussion

closed 2025-09-24 by alisterburt