warpem/warp #374: ts_export_particles: RELION5 "ERROR: MetaDataTable::read: File extracted_bin6_box54.star does not exist"

TaminoCairoli, 2025-05-04:
Dear Warp Team,

I’d appreciate some advice on organising my directories and file naming conventions, as I suspect a misstep on my part might be causing some integration issues.

I successfully pre-processed and denoised my tomograms in Warp, and picking in TomoSlice also worked well. However, during particle extraction in Warp, I had to manually rename the denoised tomogram in the relion5_poses file to XXX.ts.tomostar to proceed.

Warp then extracted the particles and produced the expected four output files. However, when I try to use the _optimization_set.star file in RELION 5, I get the following error:

`ERROR: MetaDataTable::read: File extracted_bin6_box54.star does not exist`

I can bypass this by manually editing the .star files (*_optimization_set.star and *_tomograms.star) and inserting the full path for each file that it calls (e.g.: dummy_tiltseries.mrc). This helps initially, but then I encounter another error during processing:

`ERROR: Cannot read file ../warp_tiltseries/particleseries/WT_nasal_6w_01.st/WT_nasal_6w_01.st_27.06A_000001.mrcs It does not exist`

I tried exporting particles both with and without the --relative-output-paths flag, but the problem persists.

Here’s how my directory is currently organized:

```
Human_WT_NoATP/
├── frames/
├── gainref/
├── mdoc/
├── warp/          # All Warp & TomoSlice steps, including particle extraction
└── relion/        # Intended location for initial STA
```

Do you have any suggestions for how I might better organize my directories or file naming to avoid these path issues?

Thank you for your time and help!

Best regards,
Tamino

alisterburt, 2025-05-04:
Hi @TaminoCairoli - thanks for clearly outlining your issue!

The only issue you're having that's expected is needing to change the tomostar names in your star file... your directory structure looks as expected

Are you changing into the RELION directory before starting the RELION GUI?

If that's not it, I'd recommend 
- checking that running the tutorial doesn't cause the same issues for you after particle export
- comparing files carefully after particle export between the tutorial and your own data to identify differences 

I've never had these issues with the tutorial and I'm glad you're having success with tomoslice :-) 

Good luck!

TaminoCairoli, 2025-05-04:
Hey @alisterburt :D Thank you so much for your quick reply!

Very good input with the tutorial data, I re-processed the entire tutorial data again and did particle picking in tomoslice as well! (TBH, I used some random paths instead of sphere annotation, because I did not care about the particle information now). 

Unfortunately, I also get the same error. 
`ERROR: MetaDataTable::read: File extracted.star does not exist`

To your question: Yes, I cd into the relion directory and open a new RELION project directory there. 

Just very briefly my workflow to double check I did it as it is supposed to be: 
[pre-processing in warp] -> Noise2Map -> Pick denoised tomograms with tomoslice -> generate poses -> convert poses (relion5) -> extract particles in warp -> 3D Classification/3D Refinment in RELION with optimisation_set.star

Do you think this might be an error from RELION's side that it is not recognising the files? Or rather - that it only recognises the files after providing the full path?

sorry for the inconvenience!

alisterburt, 2025-05-04:
@TaminoCairoli can you instead do the template matching and export particles based on the template matching output? This will isolate whether the problem is with the input files from tomoslice or RELION

TaminoCairoli, 2025-05-04:
Hey @alisterburt I tried with the template matching particles. Still the same error, so I guess the problem is on RELION's side? Or you think it could be some issue with our HPC?

`ERROR: MetaDataTable::read: File matching.star does not exist`

thanks!

TaminoCairoli, 2025-05-04:
Hey @alisterburt - The problem seems to be fixed. I created an own "relion" directory and tried to launch the GUI from there. But this was NOT the same relion directory as created during particle extraction. Now that I am running RELION5 from the relion directory (where also the four particle extraction files are created) RELION5 does not show the error anymore. 

Sorry for your troubles and thousand thanks for your help! I really drew a blank here :D 

Thanks again!!

closed 2025-05-04 by alisterburt