warpem/warp #378: Using extracted particles from Warp to produce an initial reference in Relion 5 (Reconstruct Particle)

mpm896, 2025-05-08:
Hello, I'm curious if anyone has used a similar workflow. I refined a structure in Dynamo at bin 4 dimensions, converted the Dynamo table to a starfile with `dynamo2warp` and re-extracted particles with WarpTools (2d) at unbinned dimensions. Rather than re-converting everything to Dynamo conventions to produce an initial reference at unbinned dimensions to then convert back to use in Relion, I figured I would do this in Relion using the `Reconstruct Particle` job (seemed easy enough anyways). However I'm running into issues with the tomogram star file - First it was that `_rlnTomoTiltSeriesStarFile #14` wasn't in the `tomograms.star` file (which I could easily add), but it seems that the `tomostar` files don't contain the right fields. Just curious if anyone has worked out a similar workflow!

alisterburt, 2025-05-08:
Hi @mpm896 - there's no way to do this directly from 2D particles but it's easy enough to do with 3D, just extract the 3D particles with the same settings then run relion_reconstruct with --3d_rot and --ctf

mpm896, 2025-05-09:
Hey @alisterburt thanks, that seems straightforward enough. I went to re-extract 3D particles with WarpTools but am now running into a cuFFT runtime error:

`WarpTools ts_export_particles --settings warp_tiltseries.settings --input_directory particlestar/ --coords_angpix 7.904 --output_star relion/run5_eo_3d.star --output_angpix 1.976 --box 128 --diameter 100 --relative_output_paths --3d`

Output:
```
Running command ts_export_particles with:
input_star = null
input_directory = particlestar/
input_pattern = *.star
coords_angpix = 7.904
normalized_coords = False
output_star = relion/run5_eo_3d.star
output_angpix = 1.976
box = 128
diameter = 100
relative_output_paths = True
2d = False
3d = True
dont_normalize_input = False
dont_normalize_3d = False
n_tilts = null
device_list = {  }
perdevice = 1
workers = {  }
settings = warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /common/workdir/MM038_1_Tau/warp/tomostar
16 files found
Parsing previous results for each item, if available...
16/16, previous metadata found for 16                                              
Found 1 files in particlestar/ matching *.star;
Found 9458 particles in 6 tilt series
Connecting to workers...
Connected to 1 workers
0/16terminate called after throwing an instance of 'std::runtime_error'
  what():  cuFFT error: CUFFT_EXEC_FAILED at /usr/share/miniconda/envs/package-build/conda-bld/warp_1720036028036/work/NativeAcceleration/gtom/src/FFT/FFT.cu:40

Failed to process /common/workdir/MM038_1_Tau/warp/warp_tiltseries/Position_5_5_ali.tomostar, marked as unselected
Check logs in /common/workdir/MM038_1_Tau/warp/warp_tiltseries/logs for more info.
Use the change_selection WarpTool to reactivate this item if required.
Unhandled exception. System.AggregateException: One or more errors occurred. (Cannot assign requested address (localhost:44141))
 ---> System.Net.Http.HttpRequestException: Cannot assign requested address (localhost:44141)
 ---> System.Net.Sockets.SocketException (99): Cannot assign requested address
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError error, CancellationToken cancellationToken)
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at System.Net.Sockets.Socket.<ConnectAsync>g__WaitForConnectWithCancellation|285_0(AwaitableSocketAsyncEventArgs saea, ValueTask connectTask, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationTo
ken)
   at System.Net.Http.HttpConnectionPool.ConnectAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.CreateHttp11ConnectionAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.AddHttp11ConnectionAsync(QueueItem queueItem)
   at System.Threading.Tasks.TaskCompletionSourceWithCancellation`1.WaitWithCancellationAsync(CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.SendWithVersionDetectionAndRetryAsync(HttpRequestMessage request, Boolean async, Boolean doRequestAuth, CancellationToken cancell
s, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at Warp.WorkerConsole.SetFileOutput(String path) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1720036028036/work/WarpLib/WorkerWrapper.cs:line 550
   at WarpTools.Commands.BaseCommand.<>c__DisplayClass1_0.<IterateOverItems>b__0(Int32 iitem, Int32 threadID) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1720036028036/work/WarpTools/Commands/BaseCommand.cs:line 105
   at Warp.Tools.Helper.ForCPUGreedy(Int32 fromInclusive, Int32 toExclusive, Int32 nThreads, Action`1 funcSetup, Action`2 funcIterator, Action`1 funcTeardown) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1720036028036/work/WarpLib/Tools/Helper.cs:line 786
at WarpTools.Commands.BaseCommand.IterateOverItems(WorkerWrapper[] workers, BaseOptions cli, Action`2 body, Int32 oversubscribe) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1720036028036/work/WarpTools/Commands/BaseCommand.cs:line 67
   at WarpTools.Commands.ExportParticlesTiltseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1720036028036/work/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 186
   at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1720036028036/work/WarpTools/Program.cs:line 30
[... cut ...]
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted
```

This is the third time in a row it's happened, and each time has happened at a different tomogram. There are no other processes running and the contents of the log file are:
`2025-05-09 02:01:31.895 Received "TomoExportParticleSubtomos", with 4 arguments, for GPU #0, 1267 MB free:`

Not really sure what's going on here! I had previously exported 3D particles but must have accidentally deleted its star file when starting with relion and had no issues.

alisterburt, 2025-05-09:
That's not enough memory on the GPU, only 1GB free... are you sure there's nothing else running?

mpm896, 2025-05-09:
Very good point... `nvidia-smi` showed no running processes but 23/24GB memory used, turns out a crashed relion_refine run from yesterday was still eating up the GPU memory.

I got the initial unbinned reference with
 `mpirun -np 4 relion_reconstruct_mpi --i run5_eo_3d.star --o initial_ref_unbinned.mrc --ctf --3d_rot`
but I was wondering if there was a way to do this from within the gui? Not that this is difficult to write out as a command, but I was very confused trying to figure out how to run this from the gui. I understand this isn't a warp issue anymore so apologies for continuing to ask about relion!

alisterburt, 2025-05-09:
No way to do it from the gui!

closed 2025-05-09 by alisterburt