warpem/warp #379: Export gimbal locked particles leads to _rlnAngle columns not being present in the output star

rkjensen, 2025-05-12:
When you export particles (WarpTools export_particles) from a star file that has one or more particles which are gimbal locked (i.e. _rlnAngleTilt = 0 or 180), warp silently just removes the _rlnAngle columns from the exported star file.

It would be useful to get a loud warning when this happens.

alisterburt, 2025-05-12:
thanks for the report Rasmus - I had another similar report recently, will look into it

alisterburt, 2025-05-12:
@rkjensen can't reproduce

input star:
```txt
data_

loop_
_rlnCoordinateX #1
_rlnCoordinateY #2
_rlnCoordinateZ #3
_rlnAngleRot #4
_rlnAngleTilt #5
_rlnAnglePsi #6
_rlnMicrographName #7
_rlnAutopickFigureOfMerit #8
   0.84195405   0.08227848   0.1923077        67.5  0        240  TS_1.tomostar  6.0833936
   0.85057473    0.2721519  0.30769232   112.49999  0      187.5  TS_1.tomostar   7.375664
    0.9051724   0.30590716  0.30769232    74.99999  180       22.5  TS_1.tomostar  6.2018175
   0.87931037   0.14135021  0.32051283    74.99999  180        180  TS_1.tomostar  6.4757905
    0.8965517    0.2763713  0.32051283        67.5  0.0         45  TS_1.tomostar  6.1560464
   0.85057473    0.3607595  0.32051283        67.5  180.0  37.499996  TS_1.tomostar  7.6919837
```

2D output:
```txt

data_general

_rlnTomoSubTomosAre2DStacks   1

data_optics

loop_
_rlnOpticsGroup #1
_rlnOpticsGroupName #2
_rlnSphericalAberration #3
_rlnVoltage #4
_rlnTomoTiltSeriesPixelSize #5
_rlnCtfDataAreCtfPremultiplied #6
_rlnImageDimensionality #7
_rlnTomoSubtomogramBinning #8
_rlnImagePixelSize #9
_rlnImageSize #10
_rlnAmplitudeContrast #11
  1  opticsGroup1  2.700  300.000  0.78940  1  2  10.13428  8.00000  32  0.070

data_particles

loop_
_rlnTomoName #1
_rlnTomoParticleId #2
_rlnCoordinateX #3
_rlnCoordinateY #4
_rlnCoordinateZ #5
_rlnAngleRot #6
_rlnAngleTilt #7
_rlnAnglePsi #8
_rlnTomoParticleName #9
_rlnOpticsGroup #10
_rlnImageName #11
_rlnOriginXAngst #12
_rlnOriginYAngst #13
_rlnOriginZAngst #14
_rlnTomoVisibleFrames #15
  TS_1.tomostar  1  365.468   48.705  18.957   67.500    0.000  240.000  TS_1/1  1  particleseries/TS_1/TS_1_8.00A_000001.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
  TS_1.tomostar  2  369.210  161.101  30.331  112.500    0.000  187.500  TS_1/2  1  particleseries/TS_1/TS_1_8.00A_000002.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
  TS_1.tomostar  3  392.909  181.082  30.331   75.000  180.000   22.500  TS_1/3  1  particleseries/TS_1/TS_1_8.00A_000003.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
  TS_1.tomostar  4  381.683   83.672  31.595   75.000  180.000  180.000  TS_1/4  1  particleseries/TS_1/TS_1_8.00A_000004.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
  TS_1.tomostar  5  389.167  163.598  31.595   67.500    0.000   45.000  TS_1/5  1  particleseries/TS_1/TS_1_8.00A_000005.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
  TS_1.tomostar  6  369.210  213.552  31.595   67.500  180.000   37.500  TS_1/6  1  particleseries/TS_1/TS_1_8.00A_000006.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
```

3d output:

```txt

data_

loop_
_rlnCoordinateX #1
_rlnCoordinateY #2
_rlnCoordinateZ #3
_rlnAngleRot #4
_rlnAngleTilt #5
_rlnAnglePsi #6
_rlnMicrographName #7
_rlnMagnification #8
_rlnDetectorPixelSize #9
_rlnCtfMaxResolution #10
_rlnImageName #11
_rlnCtfImage #12
_rlnPixelSize #13
_rlnVoltage #14
_rlnSphericalAberration #15
  365.468   48.705  18.957       67.5    0        240  TS_1.tomostar  10000.0  8.00000  4.3  subtomo/TS_1/TS_1_0000000_8.00A.mrc  subtomo/TS_1/TS_1_0000000_ctf_8.00A.mrc  8.00000  300.000  2.700
  369.210  161.101  30.331  112.49999    0      187.5  TS_1.tomostar  10000.0  8.00000  4.3  subtomo/TS_1/TS_1_0000001_8.00A.mrc  subtomo/TS_1/TS_1_0000001_ctf_8.00A.mrc  8.00000  300.000  2.700
  392.909  181.082  30.331   74.99999  180       22.5  TS_1.tomostar  10000.0  8.00000  4.3  subtomo/TS_1/TS_1_0000002_8.00A.mrc  subtomo/TS_1/TS_1_0000002_ctf_8.00A.mrc  8.00000  300.000  2.700
  381.683   83.672  31.595   74.99999  180        180  TS_1.tomostar  10000.0  8.00000  4.3  subtomo/TS_1/TS_1_0000003_8.00A.mrc  subtomo/TS_1/TS_1_0000003_ctf_8.00A.mrc  8.00000  300.000  2.700
  389.167  163.598  31.595       67.5    0         45  TS_1.tomostar  10000.0  8.00000  4.3  subtomo/TS_1/TS_1_0000004_8.00A.mrc  subtomo/TS_1/TS_1_0000004_ctf_8.00A.mrc  8.00000  300.000  2.700
  369.210  213.552  31.595       67.5  180  37.499996  TS_1.tomostar  10000.0  8.00000  4.3  subtomo/TS_1/TS_1_0000005_8.00A.mrc  subtomo/TS_1/TS_1_0000005_ctf_8.00A.mrc  8.00000  300.000  2.700
```

alisterburt, 2025-05-12:
closing but feel free to reopen if necessary 🙂

rkjensen, 2025-05-15:
@alisterburt 

Thank you for checking. I finally had time to take a look again, and I see that the issue is not actually the gimbal lock as I first thoguht but when a particle has all euler angles being 0

I.e. if I extract from the following star file it works:
```
data_

loop_
_rlnCoordinateX #1
_rlnCoordinateY #2
_rlnCoordinateZ #3
_rlnAngleRot #4
_rlnAngleTilt #5
_rlnAnglePsi #6
_rlnMicrographName #7
_rlnClassNumber #8
13.0    40.0    90.0    1.0     0.0     0.0     00032.tomostar  1
```

But not if I extract from the following:
```

data_

loop_
_rlnCoordinateX #1
_rlnCoordinateY #2
_rlnCoordinateZ #3
_rlnAngleRot #4
_rlnAngleTilt #5
_rlnAnglePsi #6
_rlnMicrographName #7
_rlnClassNumber #8
13.0    40.0    90.0    0.0     0.0     0.0     00032.tomostar  1
```

alisterburt, 2025-05-15:
ah, this makes sense - I have some code which grabs the euler angles from a file and returns all zeros if no angles are found

If no angles are found, it doesn't write eulers into the output

https://github.com/warpem/warp/blob/7c5e5c622d9e97c91da558d88bb2b9f341c3bf39/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs#L653-L662

https://github.com/warpem/warp/blob/7c5e5c622d9e97c91da558d88bb2b9f341c3bf39/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs#L185

https://github.com/warpem/warp/blob/7c5e5c622d9e97c91da558d88bb2b9f341c3bf39/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs#L844-L849

Did this cause you some problem or was it just unexpected? All zero eulers is ~0 information so removing it doesn't seem harmful but I see why it would be unexpected...