warpem/warp #380: Euler Angle convenions Warp vs RELION5

TaminoCairoli, 2025-05-13:
Dear Warp team!

I believe to have an issue with the behaviour with my particles in my Warp-RELION5 pipeline. 

Short background to the workflow: Pre-processing in Warp, particle picking in tomoslice and refinement at bin6 and bin3 in RELION5. Next, I classified my particles.

I realized that the classification differs significantly in the quality, depending if I provide the "relion_optimisation_set.star" file from my most recent refinement, or if I input the same particles but re-extracted in Warp. The warp output is better.  

Another thing I noticed: 
When I do classification in RELION5 and sub-select one class, I obtain a particle.star file. If I extract this particle.star file again in warp (same binning) and subject it to a 3D classification with 1 iteration, 1 class and no alignment in RELION5, the output looks randomized (see attached files). The output also changes drastically if I provide a reference structure or not. 

Additionally, in all my RELON5 jobs I observe a warning message. It is not fatal for the job: 
`WARNING: tomogram WT_nasal_6w_08.st.tomostar has relion-4 definition of projection matrices; converting them now... `

Selected class in RELION5
[image]

Particle .star file from selected class re-extracted in Warp and run through 3D classification with 1 iteration and 1 class (no alignment). 
[image]

I am not sure if I am doing something wrong from my side, please let me know :) 

Thanks, 

Tamino

alisterburt, 2025-05-13:
Hi @TaminoCairoli - great to see you here 🙂 

> I realized that the classification differs significantly in the quality, depending if I provide the "relion_optimisation_set.star" file from my most recent refinement, or if I input the same particles but re-extracted in Warp. The warp output is better.

The only thing I can think of that explains this would be updated CTF parameters which are correct at the refined particle center... are there large shifts during your first refinements?

> When I do classification in RELION5 and sub-select one class

What exactly does this operation do to the STAR files? You stated above that you have been able to successfully extract some particles after running RELION, right? My guess is that the subset selection job is doing something weird but I don't know...

> Additionally, in all my RELON5 jobs I observe a warning message. It is not fatal for the job:
WARNING: tomogram WT_nasal_6w_08.st.tomostar has relion-4 definition of projection matrices; converting them now... 

This isn't an issue and unfortunately we can't easily fix it, the RELION metadata model can subtly incompatible with the Warp/M metadata model depending on how tilt angles are handled in Warp

alisterburt, 2025-05-13:
huh weird - so 3D classification is definitely not the most reproducible process, if I only saw this once I wouldn't be 100% sure that this different for Warp vs. RELION, especially because your particles came from Warp in both places... there might also be some bug where Class3D without alignments doesn't take into account the alignments in a file properly, just an idea, haven't confirmed...

I would try to write a script for doing the subset selection yourself to make sure nothing weird is going on there
https://teamtomo.org/starfile/ should make it easy, see https://gist.github.com/alisterburt/8f8858dbbb4bc11f62b21df34e3cbeee for an example of doing stuff with RELION STAR files

TaminoCairoli, 2025-05-14:
Hey @alisterburt! 

It's true that it's not always super reproducible, I will have an eye on that for future datasets. Thanks for the links, they are helpful! I will play with it and find a work-around :)

Thanks for your time and insights!!

All the best, 
Tamino

alisterburt, 2025-09-24:
going to close this as it seems to be a RELION subset selection issue but feel free to reopen if you think this is something we need to look at

closed 2025-09-24 by alisterburt