warpem/warp #381: Euler Angles in star files from ts_export_particles --2d

mpm896, 2025-05-13:
Hello,

I've been having trouble producing a sensible reconstruction from Relion 3D classification or 3D auto-refine. I'm trying to resolve a filament structure, and aligned everything in Dynamo at bin 4. After extracting 3D particles, I can reproduce an initial reference with `relion_reconstruct --i run5_eo_3d.star --o initial_ref_unbinned.mrc --ctf --3d_rot`, so the angles must have converted fine from Dynamo using `dynamowarp`. However when exporting 2d particle stacks and inputting the `_optimisation_set.star` file into the Class3D or 3D auto-refine job, I cannot produce anything that resembles a filament. I noticed in the 2d particle star file that the Euler Angles are wildly different than the 3d particle star file. Is this expected? Thanks!

mpm896, 2025-05-13:
Just noticed #380 and thought these might be related issues. I'm also adding the first 5 particle entries from the 2d and 3d particle star files here to show the differences:

2d:
```
data_particles

loop_
_rlnTomoName #1
_rlnTomoParticleId #2
_rlnCoordinateX #3
_rlnCoordinateY #4
_rlnCoordinateZ #5
_rlnAngleRot #6
_rlnAngleTilt #7
_rlnAnglePsi #8
_rlnTomoParticleName #9
_rlnOpticsGroup #10
_rlnImageName #11
_rlnOriginXAngst #12
_rlnOriginYAngst #13
_rlnOriginZAngst #14
_rlnTomoVisibleFrames #15
Position_5_5_ali.tomostar     1  448.500  498.500  121.500    3779.229  5554.653  -3042.979     Position_5_5_ali/1  1  ../warp_tiltseries/particleseries/Position_5_5_ali/Position_5_5_ali_7.90A_000001.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
  Position_5_5_ali.tomostar     2  446.500  497.500  121.500    3000.580  4795.943   -990.644     Position_5_5_ali/2  1  ../warp_tiltseries/particleseries/Position_5_5_ali/Position_5_5_ali_7.90A_000002.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
  Position_5_5_ali.tomostar     3  444.500  496.500  121.500    2589.196  4578.907  -1010.697     Position_5_5_ali/3  1  ../warp_tiltseries/particleseries/Position_5_5_ali/Position_5_5_ali_7.90A_000003.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
  Position_5_5_ali.tomostar     4  443.500  495.500  121.500    4673.043  5138.858  -1012.416     Position_5_5_ali/4  1  ../warp_tiltseries/particleseries/Position_5_5_ali/Position_5_5_ali_7.90A_000004.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
  Position_5_5_ali.tomostar     5  441.500  494.500  121.500    3818.190  4795.943   -991.217     Position_5_5_ali/5  1  ../warp_tiltseries/particleseries/Position_5_5_ali/Position_5_5_ali_7.90A_000005.mrcs  0.0  0.0  0.0  [1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]
```

3d:
```
data_

loop_
_rlnCoordinateX #1
_rlnCoordinateY #2
_rlnCoordinateZ #3
_rlnAngleRot #4
_rlnAngleTilt #5
_rlnAnglePsi #6
_rlnMicrographName #7
_rlnMagnification #8
_rlnDetectorPixelSize #9
_rlnCtfMaxResolution #10
_rlnImageName #11
_rlnCtfImage #12
_rlnPixelSize #13
_rlnVoltage #14
_rlnSphericalAberration #15
448.500  498.500  121.500       65.96  96.947    -53.11  Position_5_5_ali.tomostar  10000.0  7.90400  5.6  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000000_7.90A.mrc  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000000_ctf_7.90A.mrc  7.90400  200.000  2.700
  446.500  497.500  121.500       52.37  83.705    -17.29  Position_5_5_ali.tomostar  10000.0  7.90400  5.6  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000001_7.90A.mrc  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000001_ctf_7.90A.mrc  7.90400  200.000  2.700
  444.500  496.500  121.500       45.19  79.917    -17.64  Position_5_5_ali.tomostar  10000.0  7.90400  5.6  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000002_7.90A.mrc  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000002_ctf_7.90A.mrc  7.90400  200.000  2.700
  443.500  495.500  121.500       81.56   89.69    -17.67  Position_5_5_ali.tomostar  10000.0  7.90400  5.6  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000003_7.90A.mrc  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000003_ctf_7.90A.mrc  7.90400  200.000  2.700
  441.500  494.500  121.500       66.64  83.705     -17.3  Position_5_5_ali.tomostar  10000.0  7.90400  5.6  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000004_7.90A.mrc  ../warp_tiltseries/subtomo/Position_5_5_ali/Position_5_5_ali_0000004_ctf_7.90A.mrc  7.90400  200.000  2.700
```

mpm896, 2025-05-13:
Just found #227 -- I'm currently running a version of WarpTools that I installed in July 2024, before this patch. I'm running with the 3d particles as input and it's looking like exactly what I would expect. Hopefully this explains the issue!

alisterburt, 2025-05-13:
@mpm896 glad you found #227 - definitely good to keep updated as there have been and likely will continue to be bugfixes 🙂

alisterburt, 2025-05-13:
gonna close this as it looks like you're sorted

closed 2025-05-13 by alisterburt