warpem/warp #383: 3D Reconstruction Appears Fragmented/Granular in M Compared to Reference (EMPIAR-10499)

xinsheng44, 2025-05-19:
Hi,
I'm encountering an issue during 3D reconstruction using M on the EMPIAR-10499 dataset. The result appears fragmented or granular, as shown in Figure 1. However, the official result generated using M (presumably with the recommended pipeline) produces a clean and coherent 3D structure, shown in Figure 2.

Figure 1
[image]

Figure 2
[image]

My question is:
Which parameters are most likely to cause the reconstruction to become fragmented or granular like in Figure 1?

Are there any recommended settings (e.g., --lowpass, --diameter, --mask, or pixel size options) that could help avoid over-smoothing or over-filtering, which may be responsible for the loss of structural integrity?

Any advice or insight into what may be going wrong—or how I can better reproduce the high-quality result from the EMPIAR-10499 official reconstruction—would be greatly appreciated.

Thanks in advance for your help!

To investigate, I followed the STA workflow and used the following command to create the species:

`
GPU="1"
PORT="14488"
# 添加超参数
WARP_GRID="16x12"  # refine_imagewarp 的默认值
LOWPASS="20"
WORK_DIR="m_${WARP_GRID}_${LOWPASS}"  # 基于 WARP_GRID 创建工作目录

# 步骤35: 使用M创建种群
run_step 35 "使用M创建种群" MTools create_population \
    --directory $WORK_DIR \
    --name 10499

# 步骤36: 创建源
run_step 36 "创建源" MTools create_source \
    --name 10499 \
    --population $WORK_DIR/10499.population \
    --processing_settings warp_tiltseries.settings

# 步骤37: 创建掩码
run_step 37 "创建掩码" bash -c "cd relion && relion_mask_create \
    --i Refine3D/job004/run_class001.mrc \
    --o ../$WORK_DIR/mask_1.7apx.mrc \
    --ini_threshold 0.00325"

# 步骤38: 创建物种
run_step 38 "创建物种" MTools create_species \
    --population $WORK_DIR/10499.population \
    --name ribosome_70S \
    --diameter 220 \
    --sym C1 \
    --temporal_samples 1 \
    --half1 ./relion/Refine3D/job004/run_half1_class001_unfil.mrc \
    --half2 ./relion/Refine3D/job004/run_half2_class001_unfil.mrc \
    --mask $WORK_DIR/mask_1.7apx.mrc \
    --particles_relion ./relion/Refine3D/job004/run_data.star \
    --angpix 1.7005 \
    --angpix_resample 1.7005 \
    --lowpass $LOWPASS

# 步骤39: 初始化MCore
run_step 39 "初始化MCore" MCore \
    --port $PORT \
    --population $WORK_DIR/10499.population \
    --iter 0 \
    --devicelist $GPU

# 步骤40: 第一次MCore精炼
run_step 40 "第一次MCore精炼" MCore \
    --port $PORT \
    --population $WORK_DIR/10499.population \
    --refine_imagewarp $WARP_GRID \
    --refine_particles \
    --ctf_defocus \
    --ctf_defocusexhaustive \
    --perdevice_refine 2 \
    --devicelist $GPU

# 步骤41: 第二次MCore精炼
run_step 41 "第二次MCore精炼" MCore \
    --port $PORT \
    --population $WORK_DIR/10499.population \
    --refine_imagewarp $WARP_GRID \
    --refine_particles \
    --ctf_defocus

# 步骤42: 第三次MCore精炼
run_step 42 "第三次MCore精炼" MCore \
    --port $PORT \
    --population $WORK_DIR/10499.population \
    --refine_imagewarp $WARP_GRID \
    --refine_particles \
    --refine_stageangles

# 步骤43: 第四次MCore精炼
run_step 43 "第四次MCore精炼" MCore \
    --port $PORT \
    --population $WORK_DIR/10499.population \
    --refine_imagewarp $WARP_GRID \
    --refine_particles \
    --refine_mag \
    --ctf_cs \
    --ctf_defocus \
    --ctf_zernike3

# 步骤44: 估计权重(项目)
run_step 44 "估计权重(项目)" EstimateWeights \
    --port $PORT \
    --population $WORK_DIR/10499.population \
    --source 10499 \
    --resolve_items

# 步骤45: MCore处理
run_step 45 "MCore处理" MCore \
    --port $PORT \
    --population $WORK_DIR/10499.population

# 步骤46: 估计权重(帧)
run_step 46 "估计权重(帧)" EstimateWeights \
    --population $WORK_DIR/10499.population \
    --source 10499 \
    --resolve_frames

# 步骤47: MCore粒子精炼
run_step 47 "MCore粒子精炼" MCore \
    --port $PORT \
    --population $WORK_DIR/10499.population \
    --perdevice_refine 4 \
    --refine_particles

步骤48: 重采样轨迹
run_step 48 "重采样轨迹" MTools resample_trajectories \
    --population $WORK_DIR/10499.population \
    --species $WORK_DIR/species/ribosome_70S_e9640476/ribosome_70S.species \
    --samples 2

# 步骤49: 最终MCore精炼
run_step 49 "最终MCore精炼" MCore \
    --port $PORT \
    --population $WORK_DIR/10499.population \
    --refine_imagewarp $WARP_GRID \
    --refine_particles \
    --refine_stageangles \
    --refine_mag \
    --ctf_cs \
    --ctf_defocus \
    --ctf_zernike3 \
    --devicelist $GPU

# 步骤50: 创建PostProcess/job002目录
run_step 50 "创建PostProcess/job002目录" bash -c "cd relion && mkdir -p PostProcess/job002"

# 步骤51: 最终后处理
run_step 51 "最终后处理" bash -c "cd relion && \$(which relion_postprocess) \
    --mask ../$WORK_DIR/species/ribosome_70S_e9640476/ribosome_70S_mask.mrc \
    --i ../$WORK_DIR/species/ribosome_70S_e9640476/ribosome_70S_half1.mrc \
    --o PostProcess/job002/postprocess \
    --angpix -1 \
    --auto_bfac \
    --autob_lowres 10 \
    --pipeline_control PostProcess/job002/"
`

alisterburt, 2025-05-19:
Hi @xinsheng44 - your warp grid parameter is set extremely high and your model will be overfit as a result. 

I would revert to before having done any M processing and first start with 1x1 image warp for a few rounds to see how far this can help

xinsheng44, 2025-05-19:
Hi @alisterburt

alisterburt, 2025-05-19:
@xinsheng44 how well refinement in M works will depend much more on the quality of your initial tilt series alignments and the quality of your particle set than your specific M refinement parameters (within reason) - I suspect the first two are your limiting factors. How well did initial refinements in RELION work after particle picking?

The M result from the paper was from before Warp 2.0 was released so was not produced using exactly the workflow outlined in the tutorial, see the paper for details 🙂

closed 2025-06-13 by alisterburt