warpem/warp #394: fixing MCore crash during "Commiting changes to xxxx" step

cai-cryo, 2025-06-29:
Hi,

Regarding the error that MCore would crash during commiting changes to data source, I found a temporal solution that seems to work.

The reason for this error seems to be that MCore will try to use all available CPUs during the calculation of new Hash when commiting changes, which might cause WorkerDied. Limiting the parallel number of this step helped to solve my issue.

Changes to the code is attached below, hope this useful.

Best,
Cai

For warp-2.0.0dev33,

I changed the line355 from the code of WarpLib/Sociology/DataSource.cs from :

        public void ComputeVersionHash()
        {
            StringBuilder Builder = new StringBuilder();

            string[] Paths = Files.Values.ToArray();
            string[] Hashes = new string[Paths.Length];
            Parallel.For(0, Paths.Length, (i) =>
            {
                var file = Paths[i];
                Movie Movie = IsTiltSeries ? new TiltSeries(FolderPath + file) : new Movie(FolderPath + file);
                Hashes[i] = Movie.GetProcessingHash();
            });

            Builder.AppendJoin("", Hashes);

            foreach (var species in UsedSpecies)
            {
                Builder.Append(species.Key);
                Builder.Append(species.Value);
            }

            Version = MathHelper.GetSHA1(Helper.ToBytes(Builder.ToString().ToCharArray()));
        }

to modified

        public void ComputeVersionHash()
        {
            StringBuilder Builder = new StringBuilder();

            string[] Paths = Files.Values.ToArray();
            string[] Hashes = new string[Paths.Length];

    // Define parallel options to limit concurrency
	    ParallelOptions parallelOptions = new ParallelOptions
	    {
	        MaxDegreeOfParallelism = 20 // Limit to a maximum of 20 concurrent operations
	    };

	    // Use Parallel.For with the specified parallel options
	   Parallel.For(0, Paths.Length, parallelOptions, (i) =>
	    {
	        var file = Paths[i];
	        Movie Movie = IsTiltSeries ? new TiltSeries(FolderPath + file) : new Movie(FolderPath + file);
 	       Hashes[i] = Movie.GetProcessingHash();
 	    });

            Builder.AppendJoin("", Hashes);

            foreach (var species in UsedSpecies)
            {
                Builder.Append(species.Key);
                Builder.Append(species.Value);
            }

            Version = MathHelper.GetSHA1(Helper.ToBytes(Builder.ToString().ToCharArray()));
        }

shahpnmlab, 2025-06-29:
Thanks @cai-cryo for this insightful patch. I will test this out once I have come back to the M processing part of the pipeline and let you know if this solves my issue.

cai-cryo, 2025-06-30:
Hi Alister and Dimitry,

I hope I have identified the correct issue.

We are getting great results using Warp/M, further credit will not be necessary for me.

Best regard,
Cai Bin

PR #395 (2025-06-30): explicitly set lower degree of parallelism when computing species hash 
closes #394  - thank you @cai-cryo for raising the issue and providing your solution!

This PR only changes behavior in environments with large numbers of CPUs