warpem/warp #412: Ability to exclude specific frames within a tilt series, and can you resume a run where you left off if it crashes mid run?

sam-mahdi, 2025-09-13:
Is there a way to exclude specific frames when importing your tilt series prior to alignment? There are options to exclude based off angle and intensity, however sometimes you want to exclude specific frames. Additionally, I appear to frequently be coming across memory issue where my runs abort with a core dump. Is there a way to resume these runs starting off where you left off? If I try to rerun the same command that crashed, it'll start from the beginning and overwrite everything that was already written. Is there some argument that first checks the logs to see see where it should resume? Or an argument that prevents overwriting of files and skips those runs if their file already exists?

alisterburt, 2025-09-15:
@sam-mahdi hey! in future could you not mix multiple questions within a single issue? Thanks

re: unselection This is a problem with the lack of GUI unfortunately, you'll have to manually update the UnselectManual (or similar) entry in the XML metadata for the tilt series or remove lines from the tomostar file before subsequent processing to achieve what you're trying to do. We are planning to release a web based GUI which will simplify this workflow

re: resuming a run no this is not possible right now, afaik we generate enough metadata to do this with `processed_items.json` and I know @dtegunov is working on reintroducing an on the fly system right now so working in that part of the codebase... maybe he has thoughts?

alisterburt, 2025-09-16:
not a bad suggestion but another wrapper wouldn't be my preference as it further complicates setup of setting up a fully functional warptools installation - we would likely integrate something similar ourselves if we did this

closed 2025-09-24 by alisterburt