warpem/warp #414: AreTomo3 Failure "Assertion '0' failed"

sam-mahdi, 2025-09-16:
I cannot seem to get AreTomo3 to work on any of my tilts. This is my script

```
WarpTools ts_aretomo3 \
--settings /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/scripts/warp_tiltseries.settings \
--device_list 1 4
```
It seems to read/load the data with no problem but crashes with this error message:

```
Running command ts_aretomo3 with:
angpix = null
mask = False
alignz = 0
patches = 4,4
axis_iter = 0
axis_batch = 0
min_fov = 0
axis = null
delete_intermediate = False
thumbnails = False
exe = AreTomo3
device_list = { 1, 4 }
perdevice = 1
workers = {  }
settings = /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/scripts/warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null
input_norawdata = False
strict = False

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/tomostar
25 files found
Parsing previous results for each item, if available...
25/25, previous metadata found for 0                                                                                                                                                                                                                                                                  
Connecting to workers...
Connected to 2 workers
Current tilt axis angle: 88.600 °
Running AreTomo3 with final average tilt axis angle:
0/25

Failed to process /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/warp_tiltseries/433-1-tomo02.tomostar, marked as unselected                            
Check logs in /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/warp_tiltseries/logs for more info.
Use the change_selection WarpTool to reactivate this item if required.
Exception details:
System.Exception: Could not find 433-1-tomo02.xf
   at Warp.TiltSeries.ImportAlignments(ProcessingOptionsTomoImportAlignments options) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1756402617484/work/WarpLib/TiltSeries/TiltSeries.ImportAlignments.cs:line 84
   at WarpTools.Commands.AreTomo3Tiltseries.<>c__DisplayClass0_2.<Run>b__4(WorkerWrapper worker, TiltSeries t) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1756402617484/work/WarpTools/Commands/Tiltseries/AreTomo3Tiltseries.cs:line 185
   at WarpTools.Commands.BaseCommand.<>c__DisplayClass1_0`1.<IterateOverItems>b__0(WorkerWrapper worker, T[] items) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1756402617484/work/WarpTools/Commands/BaseCommand.cs:line 70
   at WarpTools.Commands.BaseCommand.<>c__DisplayClass2_0`1.<IterateOverItems>g__ProcessItem|2(Int32 index, Int32 threadID, Int32 jsonIndex) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1756402617484/work/WarpTools/Commands/BaseCommand.cs:line 192
```
I don't know what the .xf is supposed to be, there is no file I have generated up to this point that contains a .xf extension. Looking at the logs

```
GPU 0: Estimated CTFs
310 2025-09-16 16:25:54.633 #index  tilt  dfmin     dfmax    azimuth  phase  res   score   DfHand
311 2025-09-16 16:25:54.639    1   -60.00  37311.63  37865.96   25.42    0.0000  21.93   0.04553   1
312 2025-09-16 16:25:54.644    2   -58.01  27360.62  27767.10   20.86    0.0000  22.66   0.03304   1
313 2025-09-16 16:25:54.651    3   -56.00  37119.59  37671.06   16.87    0.0000  20.60   0.04165   1
314 2025-09-16 16:25:54.659    4   -54.00  39990.85  40584.98   16.87    0.0000  22.66   0.05228   1
315 2025-09-16 16:25:54.665    5   -52.01  22991.48  23333.06   18.87    0.0000  21.93   0.03290   1
316 2025-09-16 16:25:54.670    6   -50.00  36581.49  37124.97   25.42    0.0000  20.60   0.05147   1
317 2025-09-16 16:25:54.675    7   -48.01  37510.76  38068.05   25.42    0.0000  21.25   0.05647   1
318 2025-09-16 16:25:54.681    8   -46.01  22578.83  22914.27   16.87    0.0000  22.66   0.02922   1
319 2025-09-16 16:25:54.690    9   -44.00  21508.90  21828.45   25.42    0.0000  23.45   0.05299   1
320 2025-09-16 16:25:54.696   10   -42.01  38529.52  39101.95   16.87    0.0000  20.60   0.06980   1
321 2025-09-16 16:25:54.701   11   -40.01  38237.68  38805.77   19.62    0.0000  20.60   0.07776   1
[... 82 lines cut ...]
```
2025-09-16 16:28:28.223   43     23.99    -0.55      0.90
275 2025-09-16 16:28:28.235   44     25.99    -0.09      0.12
276 2025-09-16 16:28:28.246 CCufft2D::CreateForwardPlan: CUFFT_INTERNAL_ERROR
277 2025-09-16 16:28:28.256   45     27.99     1.15      6.06
278 2025-09-16 16:28:28.276   46     30.00    -0.51     -2.03
279 2025-09-16 16:28:28.284   47     32.00     2.51      9.93
280 2025-09-16 16:28:28.293   48     33.99    -2.54     -6.52
281 2025-09-16 16:28:28.300   49     36.00    -0.11      5.91
282 2025-09-16 16:28:28.311   50     37.98    -0.18      3.34
283 2025-09-16 16:28:28.322   51     39.98     1.39      6.71
284 2025-09-16 16:28:28.330   52     41.98    -0.46     -6.70
285 2025-09-16 16:28:28.341   53     43.98     0.63      7.37
286 2025-09-16 16:28:28.350   54     45.98     0.86     15.67
287 2025-09-16 16:28:28.360   55     47.98    -3.30     -6.27
288 2025-09-16 16:28:28.368   56     49.99     1.56     23.16
289 2025-09-16 16:28:28.386   57     51.98     0.39      8.11
290 2025-09-16 16:28:28.394   58     53.99    -3.37     15.91
291 2025-09-16 16:28:28.405   59     55.99   -22.80    -27.70
292 2025-09-16 16:28:28.414   60     57.99   -86.08    -75.98
293 2025-09-16 16:28:28.427   61     59.99  -178.49   -217.51
294 2025-09-16 16:28:28.440
295 2025-09-16 16:28:28.448 Error:   281.37

```

alisterburt, 2025-09-16:
looks like AT3 is crashing before writing out alignment metadata

@davidetorre99 is this interpretation correct?

davidetorre99, 2025-09-16:
Thanks for checking@sam-mahdi. I'd still first try running AreTomo3 standalone on your dataset to see whether the problem is in your AreTomo3 installation/setup or in the WarpTools wrapper.