warpem/warp #415: Warp-Relion high resolution refinements

mpm896, 2025-09-19:
Hello,

I'm currently working on a refinement using a Dynamo - Warp - Relion workflow, and I've hit Nyquist with bin 4 particles. Is there a recommended way to move into lower bins? I'm currently following [this guide](https://tomoguide.github.io/03-tutorial/05-sta-in-relion5/) for STA processing in relion, however I hit a will when going from bin 4 to unbinned. A `Reconstruct particle` job gives me a reconstructed average with no signal, because the tomogram.star file from Warp points to the dummy tiltseries. Is there a recommended way to go about this? Thanks!

Matt

alisterburt, 2025-09-19:
closing but do reopen if this doesn't solve your issue!

closed 2025-09-19 by alisterburt

mpm896, 2025-09-19:
Got it. I was wondering how much of the improvements I'm trying to achieve that's outlined the Tomo guide (CTF refinement and Bayesian polishing) could be achieved by M and vice versa. Let me ditch the Relion processing at this point and try out M! Thanks for you input!