warpem/warp #416: MCore setup check deteriorating half map quality and underestimating resolution?

mpm896, 2025-09-19:
Hello, 

I setup an M project from my Relion refinement, which estimated ~15Å (Nyquist at bin 4 pixel size). After setting up the M project, I ran the first check `MCore --population m/sample.population --iter 0` and in the end it estimates a resolution of 28Å. I checked the output maps (which I'm assuming are the files directly within `m/species/` and it seems that the half maps became incredibly noisy from this first `MCore` run which may be why the resolution was estimated to be so low.? Is this normal? I'm not sure why this would happen. I've attached before and after snapshots of the halfmaps below. Thanks for your input!

Before MCore:
[image]

After MCore check:

[image]

alisterburt, 2025-09-19:
hmm, actually given that no refinement is happening I would expect a much worse reconstruction if things are off... maybe give refining with 1x1 image warp + particle poses a go and see what happens!

mpm896, 2025-09-19:
Yea I would expect it to not even really resemble much of the initial average if something had been setup wrong, but I'll take a look to make sure. 

How can I switch to an earlier version to run the refinement on? I'm a little confused about the versions. After the first run of `MCore` without refinements my guess is that the latest version is directly within `m/species`, and that the original versions that were imported are within the `version` subdirectory? There are two different hashes in there so not so sure which is the proper version

alisterburt, 2025-09-19:
gotta put from the version you want to reset to in the right places (out ts_xml metadata in your warp processing directory and m files in m directory)

mpm896, 2025-09-19:
Sorry I apologize for my confusion with this. I moved the directory of the species version I want into `m/species`, but there are other things to move around such as the ts_xml metadata? When I tried re-running, `MCore` is still looking for the latest version which I see in the `name.population` file with the full hash

mpm896, 2025-09-22:
Update:

I found an issue on the Google group (can't link it due to it being blocked on company Wifi) about the specified particle diameter during `create_species` affecting the reported resolution. Although my particle is elongated with it's longest dimension being ~220Å, specifying the the diameter at 240Å was resulting in a lot of padding to the volumes. I created a new species with a diameter of 170 to keep the box size similar to what it already was, and the resulting resolution from the `MCore` check with no refinements was 16Å, only 1Å worse than the reported resolution from Relion. 

However when running `MCore` with refinements, I get the following error:
```
Unhandled exception. System.Runtime.InteropServices.ExternalException (0x80004005): {"type":"Processing error","title":"Command execution failed","status":500,"detail":"System.IndexOutOfRangeException: Index was outside the bounds of the array.\n   at Warp.TiltSeries.<>c__DisplayClass172_0.<PerformMultiParticleRefinement>b__19(Int32 t) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpLib/TiltSeries/TiltSeries.MPARefinement.cs:line 1249\n   at Warp.Tools.Helper.ArrayOfFunction[T](Func`2 generator, Int32 n) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpLib/Tools/Helper.cs:line 871\n   at Warp.TiltSeries.PerformMultiParticleRefinement(String workingDirectory, ProcessingOptionsMPARefine optionsMPA, Species[] allSpecies, DataSource dataSource, Image gainRef, DefectModel defectMap, Action`1 progressCallback) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpLib/TiltSeries/TiltSeries.MPARefinement.cs:line 1249\n   at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpWorker/WarpWorker.cs:line 843\n   at WarpWorker.API.ServiceController.EvaluateCommand() in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpWorker/API/Service.cs:line 47","traceId":"00-c91bb203e21100ac507243536a7d27dc-3b5edc5f0aa0d331-00"}
   at Warp.WorkerWrapper.SendCommand(NamedSerializableObject command) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpLib/WorkerWrapper.cs:line 239
   at Warp.WorkerWrapper.MPARefine(String path, String workingDirectory, ProcessingOptionsMPARefine options, DataSource source) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpLib/WorkerWrapper.cs:line 474
   at MCore.MCore.<>c__DisplayClass16_4.<DoProcessing>b__4(Int32 ifile, Int32 threadID) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/MCore/MCore.cs:line 474
   at Warp.Tools.Helper.<>c__DisplayClass51_0.<ForCPU>b__0(Object id) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpLib/Tools/Helper.cs:line 719
MCore_first_refine.sh: line 1: 31888 Aborted                 (core dumped) MCore --population m/HA.population --refine_imagewarp 6x4 --refine_particles --ctf_defocus
```

I found another similar issue on the Google group, and found in many of the logs in `refinement_temp/logs` there is a `without heartbeat` error, and then one log in particular contains the following error:

```
2025-09-22 16:31:13.853 Received "MPARefine", with 4 arguments, for GPU #2, 22016 MB free:
2025-09-22 16:31:13.961 Loading tilt series and masks...
2025-09-22 16:31:20.005 19384 MB after loading raw data of Position_2_3_ali.tomostar
2025-09-22 16:31:20.023 Calculating spectral weights...
2025-09-22 16:31:20.160 19378 MB after spectra estimation of Position_2_3_ali.tomostar
2025-09-22 16:31:20.234 System.IndexOutOfRangeException: Index was outside the bounds of the array.
2025-09-22 16:31:20.234    at Warp.TiltSeries.<>c__DisplayClass172_0.<PerformMultiParticleRefinement>b__19(Int32 t) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpLib/TiltSeries/TiltSeries.MPARefinement.cs:line 1249
2025-09-22 16:31:20.234    at Warp.Tools.Helper.ArrayOfFunction[T](Func`2 generator, Int32 n) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpLib/Tools/Helper.cs:line 871
2025-09-22 16:31:20.234    at Warp.TiltSeries.PerformMultiParticleRefinement(String workingDirectory, ProcessingOptionsMPARefine optionsMPA, Species[] allSpecies, DataSource dataSource, Image gainRef, DefectModel defectMap, Action`1 progressCallback) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpLib/TiltSeries/TiltSeries.MPARefinement.cs:line 1249
2025-09-22 16:31:20.234    at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /home/runner/micromamba/envs/package-build/conda-bld/warp_1750964440471/work/WarpWorker/WarpWorker.cs:line 843
2025-09-22 16:31:29.840 10 seconds without heartbeat, exiting
```

Which I guess could be from failed CTF estimation? Anyways, I'll go through the metadata very carefully to check why this kind of error would be thrown. Let me know if you have any pointers. Thanks!

alisterburt, 2025-09-22:
Hey! Sorry for the lac of docs around this process - reverting to previous M states is trivial in our (unreleased) web app and the CLI is just supposed to be a stopgap, I'm not immediately sure of exactly what needs doing so you'll have to experiment but from memory replacing all files always worked in my hands

Seems like you're on the right path with the CTF metadata - I hope you find the source and a way to avoid it by excluding certain ts, we haven't root caused this issue properly so far

alisterburt, 2025-09-22:
interesting that reported resolution varies so much with specified particle diameter cc @dtegunov this feels surprising