warpem/warp #417: Error executing align.com

sam-mahdi, 2025-09-24:
Since I was having difficulty with AreTomo, I tried to switch over to Imod, but I've been having similar issues here. 

With this input file:
```
WarpTools ts_etomo_patches \
--settings /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/scripts/warp_tiltseries.settings \
--angpix 10 \
--patch_size 500 \
--device_list 1 4

```
I got this error:
```
Connecting to workers...
Connected to 2 workers
Performing patch tracking alignment on all tilt-series...
0/25
Failed to import alignments:
Could not find 433-1-tomo02.xf
```
Checking the logs
```
2025-09-24 00:22:48.187 Batch run finished; no failures occurred
2025-09-24 00:22:53.119 Calculating projection parameters from patch tracking results in /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/warp_
2025-09-24 00:22:53.133 Running 'submfg align.com'
2025-09-24 00:22:54.472 Python PID: 1261962
2025-09-24 00:22:55.657 Running align.com ... Error executing align.com^G
2025-09-24 00:22:55.661 ERROR: tiltalign -StandardInput: exited with status 174
2025-09-24 00:23:00.817 Finished for /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/warp_tiltseries/433-1-tomo01.tomostar
2025-09-24 00:23:00.834 Execution took 834.689 seconds

```

Not sure if this is an issue with my pathway (e.g. couldn't find align.com) or it not being in my path? Not entirely sure why I'm receiving this error

alisterburt, 2025-09-24:
Do the data look okay?
Do they align outside of the warp wrapper?
Do they align manually in etomo?

sam-mahdi, 2025-09-25:
The data itself is fine, I've run AreTomo2 and etomo via RELION on it with no issues. Looking out the output files however, it seems its an environment issue:

```
IMOD_DIR (env): /programs/x86_64-linux/imod/4.12.52/IMOD
842 WARNING:
843 The environment variable IMOD_CALIB_DIRis not set.
844 Several Etomo functions will not be available:
845 Image distortion field files, Mag gradient correction, and parallel processing.
846 
847 propertyUserDir:  /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/warp_tiltseries/tiltstack/433-1-tomo01
848 GraphicsEnvironment.isHeadless()=false
849 IMOD Version: 4.12.52
850 PEET Version: 1.17.0 1-January-2024
851 
852 ETOMO_MEM_LIM=
853 MIN_AVAILABLE_MEMORY_REQUIRED=2097152.0
854 imod:  /programs/x86_64-linux/imod/4.12.52/IMOD
855 propertyUserDir:  /project/isilon/backup_4_all/backup_SMI/warp_preprocessing/warp_tiltseries/tiltstack/433-1-tomo01
856 ERROR: MONTAGESIZE - No piece list information in this image file
857 
858 ++++++++++++++++
859 Thu Sep 25 18:39:39 EDT 2025
860 433-1-tomo01
861 ++++++++++++++++
862 
863 Info:  can't open the cpu autodoc file.
864 This autodoc was not in in $IMOD_CALIB_DIR.
865 
866 Warning:Can't copy a file to itself:source:/project/isilon/backup_4_all/backup_SMI/warp_preprocessing/warp_tiltseries/tiltstack/433-1-tomo01/batchDirective.adoc,destination:/project/isilon/backup
867 java.lang.Exception: Stack trace
868         at java.lang.Thread.dumpStack(Thread.java:1336)
869         at etomo.util.Utilities.copyFile(Utilities.java:1563)
870         at etomo.util.Utilities.copyFile(Utilities.java:1545)
871         at etomo.ApplicationManager.copyDirectiveFiles(ApplicationManager.java:952)
872         at etomo.ApplicationManager.doneSetupDialog(ApplicationManager.java:505)
873         at etomo.ui.SetupReconUIHarness.doDirectiveAutomation(SetupReconUIHarness.java:132)
874         at etomo.ui.SetupReconUIHarness.doAutomation(SetupReconUIHarness.java:98)
875         at etomo.ApplicationManager.doAutomation(ApplicationManager.java:306)
876         at etomo.EtomoDirector.doAutomation(EtomoDirector.java:261)
877         at etomo.EtomoDirector.openTomogramAndDoAutomation(EtomoDirector.java:920)
878         at etomo.ui.FrontPageUIHarness.reconActionForAutomation(FrontPageUIHarness.java:44)
879         at etomo.FrontPageManager.doAutomation(FrontPageManager.java:84)
880         at etomo.EtomoDirector.doAutomation(EtomoDirector.java:239)
881         at etomo.EtomoDirector.setup(EtomoDirector.java:190)
882         at etomo.EtomoDirector.main(EtomoDirector.java:124)
883 java.lang.InterruptedException: sleep interrupted
884         at java.lang.Thread.sleep(Native Method)
885         at etomo.EtomoDirector$UtilityThread.run(EtomoDirector.java:1569)
886         at java.lang.Thread.run(Thread.java:748)
887 exitValue:0
```
I've also noticed something odd as well, in the xml file generated by warp for this, when looking at movie paths

` <MoviePath>../warp_frameseries/433-1-tomo01_30_060_-60.0.tif`
For one, there is no .tif files in the warp_frameseries directory, the motion corrected .mrc files are in the average directory. 
`../warp_frameseries/average/433-1-tomo01_30_060_-60.0.mrc`
Secondly, I've deselected a number of frames in the warp_frameseries/.xml directories, but all the frames are imported here (regardless of selection). I can see this in the .rawtlt file for etomo as well, that its reading in all the angles. I cannot see anything being excluded via the UnselectManual flag from the change_selection function. So I'm not sure where the error is coming from but there appears to be multiple issues here.

alisterburt, 2025-09-25:
too many things to respond to (on mobile)

- your IMOD installation is your responsibility, if you identify problems with Warp please open specific issues
- if having issues with our wrappers you can always run whatever program you like and import the alignments https://warpem.com/reference/warptools/custom_tilt_series_alignments/
- your note about the movie path does not seem concerning, the referenced file is the raw movie not the motion corrected average

alisterburt, 2025-09-25:
re unselect manual, did you remove data before rerunning? There might be a check for existing stacks, the flag might also be different - please experiment, I will provide a reasonable amount of help but am not responsible for your troubleshooting process