warpem/warp #418: WarpTools export rlnTomoparticlesFile missing value

mgflast, 2025-09-25:
Hello hello,

Since updating to v36 I've had the issue that the 'rlnTomoParticlesFile' field in the optimisation set .star written by WarpTools ts_export_particles is empty. Example:

```
data_

_rlnTomoParticlesFile   
_rlnTomoTomogramsFile   particles_tomograms.star
```

This despite the particle file being written (--output_star relion/tric/particle.star) and ts_export_particles running just fine. I haven't had time to look into what causes this yet, just thought I'd make this note.

Best,
Mart

alisterburt, 2025-09-25:
thanks @mgflast - cc @dtegunov not sure if you were changing things here intentionally?

dtegunov, 2025-09-25:
That code hasn't changed in a while, and I can't reproduce the error.