warpem/warp #422: warpem.com domain expired and GoDaddy advertising

rbs-sci, 2025-09-29:
Paging @dtegunov @alisterburt 

Not entirely sure where to report this, so here will do.

The warpem.com domain has expired - the one linked on the README on the main page of this project.

The most urgent point is that with the domain expired, it becomes a potential target of either (a) cybersquatting or (b) malicious actors. The first is annoying. The second a potentially nasty trap for all academics who still use (Windows) Warp.

Although the other issue is... where can Windows Warp, BoxNet data, manuals/guides etc now be accessed from?

alisterburt, 2025-09-29:
Thanks for the ping @rbs-sci - @dtegunov what do you want to do about this?

closed 2025-10-02 by alisterburt

rbs-sci, 2025-10-07:
Hi @dtegunov @alisterburt, 

Just want to give an update that this does not appear to be fully resolved yet. www.warpem.com now redirects to "expireddomains.com/domain/warpem.com"

dtegunov, 2025-10-07:
Yeah, they informed me the next day there is a transfer hold until 10/10, so we're still waiting.